espressif/idf-installer #2: esp-idf-tools-setup-online-2.3.exe 无法获取

yaozhi1234, 2021-04-12:
提供的esp-idf-tools-setup-online-2.3.exe的链接无法访问，链接提示Not Found,我应该怎么获取2.3版本，谢谢

georgik, 2021-04-12:
Thank you @yaozhi1234 for reporting the issue. Could you give us a hint where the link came from?
Download links should point to version 2.5 or higher.
Please download newer installers from page: https://dl.espressif.com/dl/esp-idf/

georgik, 2021-04-12:
@yaozhi1234 I found those links in README of the project.
These versions are deprecated and were replaced by newer one. I removed those links from documentation.
Thank you for pointing at that issue. 
Please download latest version from: https://dl.espressif.com/dl/esp-idf/

If you need an older version for some reason, let us know.

closed 2021-04-12 by georgik

yaozhi1234, 2021-04-12:
Thank you @georgik support.
esp-idf-tools-setup-online-2.3.exe download link is https://github.com/espressif/idf-installer
[image]

georgik, 2021-04-12:
@yaozhi1234 Thank you for the screenshot. The page was updated. I've added link to Download page.

commit b2c05453 (2021-04-12): fix: Closes #2, add link to installation video