espressif/idf-installer #9: Unexpected exit with ModuleNotFoundError: No module named '_socket' with installer-offline-2.6

leeebo, 2021-04-25:
1. When I try to install IDF in path `D:/esp`,  Tools path `D:/esp/.espressif`, install always failed with my PC, critical log as below:

```
2021-04-25 11:11:20.973   Log opened. (Time zone: UTC+08:00)
2021-04-25 11:11:20.973   Setup version: Inno Setup version 6.1.2
2021-04-25 11:11:20.973   Original Setup EXE: D:\Users\Administrator\Downloads\esp-idf-tools-setup-offline-2.6.exe
2021-04-25 11:11:20.973   Setup command line: /SL5="$809C4,1208102401,780800,D:\Users\Administrator\Downloads\esp-idf-tools-setup-offline-2.6.exe" 
2021-04-25 11:11:20.973   Windows version: 10.0.19042  (NT platform: Yes)
2021-04-25 11:11:20.973   64-bit Windows: Yes
2021-04-25 11:11:20.973   Processor architecture: x64
2021-04-25 11:11:20.974   User privileges: Administrative
2021-04-25 11:11:21.073   Administrative install mode: No
2021-04-25 11:11:21.073   Install mode root key: HKEY_CURRENT_USER
2021-04-25 11:11:21.073   64-bit install mode: Yes

.........................
2021-04-25 11:16:31.676   --------
2021-04-25 11:16:31.680   Resetting the repository: D:\esp\.espressif\tools\idf-git\2.30.1\cmd\git.exe -C D:\esp clean --force -d
2021-04-25 11:16:36.581   Done, exit code=0
2021-04-25 11:16:36.581   --------
2021-04-25 11:16:36.582   Running command: D:\esp\.espressif\tools\idf-git\2.30.1\cmd\git.exe -C D:\esp clean --force -d
                          ........................................
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv-20.4.0.dist-info
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/__init__.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/__main__.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/__init__.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/activator.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/bash/__init__.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/bash/activate.s
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/batch/__init__.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/batch/activate.ba
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/batch/deactivate.ba
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/batch/pydoc.ba
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/cshell/__init__.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/cshell/activate.cs
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/fish/__init__.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/fish/activate.fis
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/powershell/__init__.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/powershell/activate.ps
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/python/__init__.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/python/activate_this.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/via_template.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/xonsh/__init__.p
                          Removing .espressif/tools/idf-python/3.8.7/Lib/site-packages/virtualenv/activation/xonsh/activate.xs
                          Removing .espressif/tools/idf-python/3.8.7/Lib/s

2021-04-25 11:16:36.589   --------
2021-04-25 11:16:36.590   idf_tools.py exists in IDF directory
2021-04-25 11:16:36.590   Offline installation selected. Setting environment variable PIP_NO_INDEX=1
2021-04-25 11:16:36.590   idf_tools.py command: D:\esp\.espressif\tools\idf-python\3.8.7\python.exe "D:\esp\tools\idf_tools.py" --idf-path D:\esp
2021-04-25 11:16:36.590   Installing tools:D:\esp\.espressif\tools\idf-python\3.8.7\python.exe "D:\esp\tools\idf_tools.py" --idf-path D:\esp install
2021-04-25 11:16:36.924   Done, exit code=1
2021-04-25 11:16:36.924   --------
2021-04-25 11:16:36.924   Running command: D:\esp\.espressif\tools\idf-python\3.8.7\python.exe "D:\esp\tools\idf_tools.py" --idf-path D:\esp install
                          Traceback (most recent call last):
                            File "D:\esp\tools\idf_tools.py", line 65, in <module>
                              from urllib.request import urlretrieve
                            File "urllib\request.py", line 88, in <module>
                            File "http\client.py", line 71, in <module>
                            File "email\parser.py", line 12, in <module>
                            File "email\feedparser.py", line 27, in <module>
                            File "email\_policybase.py", line 9, in <module>
                            File "email\utils.py", line 29, in <module>
                            File "socket.py", line 49, in <module>
                          ModuleNotFoundError: No module named '_socket'
                          
                          During handling of the above exception, another exception occurred:
                          
                          Traceback (most recent call last):
                            File "D:\esp\tools\idf_tools.py", line 67, in <module>
                              from urllib import urlretrieve
                          ImportError: cannot import name 'urlretrieve' from 'urllib' (D:\esp\.espressif\tools\idf-python\3.8.7\lib\urllib\__init__.pyc)
```
2. But when I changed the Tools path to `D:/esp/.espressif` then try one more time, the installer works with no error.

Can you help to explain that?

georgik, 2021-04-26:
Thank you for reporting the problem. This is interesting behavior.

There are two directories that should be stored as separate: ESP-IDF and .espressif
- ESP-IDF contains an instance of IDF
- .espressif contains cache data and tools which can be shared between multiple instances of IDF

The observed error is caused by placing .espressif cache into the directory where ESP-IDF is being installed. In one moment installer invokes `git reset` to  invoke clean up of git directory. Git will find .espressif there and removes it. The operation will result in overall failure and the installer will roll back the whole installation.

Solution:
Install esp-idf to: D:\esp\esp-idf and .espressif cache to D:\esp\.espressif

I confirm this behavior as a bug because the installer should inform the user about the faulty configuration.

leeebo, 2021-04-26:
@georgik Thanks for the detailed explanation. The issue will be kept open until the "bug" fixed.

georgik, 2021-04-26:
@leeebo Solution is available, please test the following installer
https://github.com/espressif/idf-installer/releases/tag/online-2.8-beta-04

Installer will display warning and prompt user to choose different location in case of Tools directory is subdirectory of ESP-IDF source code repository.

Feel free to review PR #11

leeebo, 2021-04-27:
@georgik Thanks!

PR #11 (2021-04-27): tools: check whether tools directory is not under ESP-IDF source code…
Add check whether Tools are not stored under ESP-IDF source code repo.
Closes #9
closed 2021-04-27 by leeebo