espressif/idf-installer #28: Eclipse started from the last screen of the installer?

samsam4, 2021-06-28:
Could you, please, tell me where in your code you start Eclipse if I mark 'Run ESP-IDF Eclipse Environment' at the very last screen of your installer, because this is the only way when Eclipse works as supposed to work without adding extra git and python paths in the Windows Path env var on my machine with Win 10 Pro:
https://github.com/espressif/idf-eclipse-plugin/issues/283#issuecomment-869087954

Will try to investigate why your way starting Eclipse is the only way working Ok, but then next tries to start it from the desktop icon your installer created, fails unless I add paths to git and python in the Windows Path.
VSC extensions seem has [same (or similar) issue](https://github.com/espressif/vscode-esp-idf-extension/issues/434#issuecomment-869089035).
Thanks

georgik, 2021-06-28:
Hi @samsam4 . The checkbox on the last screen is basically launching just the link stored on the desktop.
https://github.com/espressif/idf-installer/blob/main/src/InnoSetup/IdfToolsSetup.iss#L228

The different behaviour might be caused by the installer. When launching from the last screen, the child process inherits environment variables from the installer all those paths including Git and Python. That's the reason why it works when launching from the last screen.

When launching from the desktop those variables are back at original value inherited from the system. So the Git and Python PATH is not set. That's the place where Eclipse plugin should read the JSON file with those values and set variables. I can imagine small workaround with a launcher which will set those variables and invoke the Eclipse. IMHO It would be better if Eclipse plugin handles this situation. What's your opinion @kolipakakondal ?

georgik, 2021-06-28:
@samsam4 Thank you for discovering this issue. Really interesting one.

Here's one possible workaround that might work:
- open ESP-IDF shell (e.g. PowerShell)
- type: `& ~\Desktop\ESP-IDF Eclipse.lnk`
This should launch the Eclipse with all variables inherited from ESP-IDF shell including correct Python and Git.

samsam4, 2021-06-28:
@georgik 
Great, thanks - it works. Can it be done similar way with cmd so I can make a script and shortcut to start a 'good' Eclipse  from a Desktop icon (I'm not quite familiar with powershell yet and not sure if I make a copy of Initialize-Idf.ps1 and add  `& '~\Desktop\ESP-IDF Eclipse.lnk' ` will it work)?
I definitely agree  that the issue should be handled from Eclipse plugin, but my posted issue for the problem stays there for a while not resolved. 
BTW this might not be even issue from the plugin itself, but some of the other code from ESP-IDF tools (idf-env.exe or so), because I have same or similar issue with the VSC extension?
Regards

kolipakakondal, 2021-06-29:
Hi @samsam4 @georgik I see a problem and found it why it's happening - It seems to be to identify the `python` version we are running a java process with the "system environment variables" which is wrong, ideally it should run based on the eclipse PATH environment variable(along with the system environment) which is already is having GIT in the PATH variable.

Let me fix this soon and provide a patch version. However, this won't impact any other functionality of the eclipse plugin other than `Help > Product Information`

Thanks @samsam4 for the logs and that helped.

samsam4, 2021-06-29:
@kolipakakondal  - I doubt the issue with paths to git and python was only with Help > Product Information. Somewhere in my previous postings there should be a log when I tried to build some example and if I don't put python in Windows Path env variable, the build process was also failing and if I remember correct - the problem was again that cant find python.
Regards

kolipakakondal, 2021-06-29:
Hi @samsam4 Ok, at least I couldn't find this info here https://github.com/espressif/idf-eclipse-plugin/issues/283. Let us know if you see a problem with a build also.

BTW, the ESP-IDF Windows installer comes with the `Eclipse plugin 2.1.0` but that's not the latest - will update this with the next version of the offline installer.

To get the latest version of `eclipse plugin 2.1.2`, please update your eclipse with the plugin update site. Please check this https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugin-using-update-site-url

samsam4, 2021-06-29:
Hi @kolipakakondal, is this version eclipse plugin 2.1.2 with fixed git and python paths or not yet?
Thanks

kolipakakondal, 2021-06-29:
Hi @samsam4 `eclipse plugin 2.1.2` does not have a fix for the git and python path issue which you've reported. It will be available in the next release.

https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.1.2

samsam4, 2021-06-29:
@kolipakakondal   - will post the other problems and error log when Eclipse cant find git and python path in the original posted issue in your repository for the Eclipse plugin, so not to pollute georgik's space as the problem is anyway not from installer...

@georgik - thank you very much again - your help so far is always competent and оn time 👍

samsam4, 2021-06-30:
@georgik  Unfortunately your hack with Power Shell that was working with v2.1.0 plugin, [doesn't work any more with newer version v2.1.2](https://github.com/espressif/idf-eclipse-plugin/issues/283#issuecomment-870724918) neither on Help>Product Information nor for setting Eclipse build environment variables. 

@kolipakakondal  Quite a regress :( :( :(

Will close this issue, as this is actually not an issue from the Installer, looks like is **solely issue with the plugin**, because VSC extension was already fixed for similar issue without any changes in IDF and tools.
Regards

closed 2021-06-30 by samsam4