espressif/idf-installer #29: Initialize-Idf.ps1 has to run in installation folder in order to work

ChrisIdema, 2021-06-28:
I want to be able to initalize esp-idf from a different path than the installation folder and without using the ESP-IDF PowerShell.
I have defined IDF_PATH in my environment variables.
Now I have to type the following command to initialize idf:
`pushd; cd $env:IDF_PATH; .$Env:IDF_TOOLS_PATH\Initialize-Idf; popd`

The following line is the problem:
https://github.com/espressif/idf-installer/blob/a8ec97cfc39db4e6b43e4b90da4c44a440e109a3/src/PowerShell/Initialize-Idf.ps1#L7

A workaround for me is to modify that line.
```
if ($null -eq $env:IDF_PATH) {
	$IDF_PATH=(Get-Location).Path
}
else{
	$IDF_PATH=$env:IDF_PATH
}
```

With my modification I only have to type:
`.$Env:IDF_TOOLS_PATH\Initialize-Idf`

I don't understand why Initialize-Idf doesn't know the installation path since this is known during installation and even written to esp_idf.json. The same applies to the bat file.

georgik, 2021-06-28:
Hi @ChrisIdema Thank you for bringing an alternative solution.

Let me add some additional facts:
The initializer does not know the location of ESP-IDF. That's true.
The initializer infers the path to ESP-IDF based on the working directory. If you're using the launch icon, it takes the working directory where .lnk is started and uses it as ESP-IDF location.
The initializer requests idf-env to retrieve values of paths (Python, Git) for the directory.
The initializer is shared between multiple instances of ESP-IDF.

Having this design allows us to have multiple installations of ESP-IDF side by side and launching them by Windows Terminal 1.7+ drop-down menu or by icons from Desktop.

Basically, you do not need to set an environment variable, just launch the initializer from the directory with ESP-IDF.

You're right that it does not allow you to initialize ESP-IDF from a different location. Your solution works well for that case. If it's necessary to switch the version of IDF, it should be sufficient to update the environment variable and call the initializer.

There is already work in progress on adding `idf-env shell` command which should bootstrap the environment from any location, the feature is not complete yet: https://github.com/espressif/idf-env/blob/main/src/shell.rs

Meanwhile, we can add your suggestion, because it does not require any complex implementation of Termial IO logic like shell mentioned above. I've added the issue to 2.10 milestone.

Thank you very much for sharing the idea.

georgik, 2021-07-29:
Available in Releases: idf-installer 2.9 beta 04.

closed 2021-07-29 by georgik

ChrisIdema, 2021-08-25:
It works for me with the latest installer. Installer 2.10, IDF v4.3. Thanks!