espressif/idf-installer #31: Initialize-Idf.ps1 relies on external software component input without validation

laszlopataki, 2021-07-14:
### Issue
Initialize-Idf.ps1 does not check the result of this query. The value comes from a different software component. 
It should be validated. 
`$IdfPythonDir=idf-env config get --property python --idf-path "$IDF_PATH\"` 

Since `idf-env config get --property python --idf-path "$IDF_PATH\"` returns nothing idf won't work. 

### Steps to reproduce: 

1. Install tools with esp-idf-tools-setup-2.3.exe. 
2. Change the Powershell shortcut on Desktop to start in a different IDF location (e.x. the idf in `git clone https://github.com/aws/amazon-freertos.git --recurse-submodules -b release freertos`)
2. Run the ESP-IDF 4.2 PowerShell

```
esp_idf.json
{
    "$schema": "http://json-schema.org/schema#",
    "$id": "http://dl.espressif.com/dl/schemas/esp_idf",
    "_comment": "Configuration file for ESP-IDF Eclipse plugin.",
    "_warning": "Use / or \\ when specifying path. Single backslash is not allowed by JSON format.",
    "gitPath": ".../.espressif/tools/idf-git/2.30.1/cmd/git.exe",
    "idfToolsPath": ".../.espressif",
    "idfSelectedId": "esp-idf-2e4624254d5bc78a121891459c52761b",
    "idfInstalled": {
        "esp-idf-2e4624254d5bc78a121891459c52761b": {
            "version": "4.2",
            "python": ".../.espressif/python_env/idf4.2_py3.8_env/Scripts/python.exe",
            "path": ".../freertos/vendors/espressif/esp-idf/"
        }
    }
}
```

### Log:

```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

Setting PYTHONNOUSERSITE, was not set
You cannot call a method on a null-valued expression.
At ...\.espressif\Initialize-Idf.ps1:52 char:1
+ $IdfPythonDir = $IdfPythonDir.Trim("`"")
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (:) [], RuntimeException
    + FullyQualifiedErrorId : InvokeMethodOnNull

Using Python in
Python 3.9.6
Using Git in C:/Users/Laszlo/.espressif/tools/idf-git/2.30.1/cmd
git version 2.30.1.windows.1
Setting IDF_PATH: ...\embedded_workspace\freertos-release\vendors\espressif\esp-idf
Adding ESP-IDF tools to PATH...

Name                           Value
----                           -----
OPENOCD_SCRIPTS                ...\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\s...
IDF_CCACHE_ENABLE              1
IDF_PYTHON_ENV_PATH           ...\.espressif\python_env\idf4.2_py3.9_env

Added to PATH
-------------
...\embedded_workspace\freertos-release\vendors\espressif\esp-idf\components\esptool_py\esptool
...\embedded_workspace\freertos-release\vendors\espressif\esp-idf\components\app_update
...\embedded_workspace\freertos-release\vendors\espressif\esp-idf\components\espcoredump
...\embedded_workspace\freertos-release\vendors\espressif\esp-idf\components\partition_table
...\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin
...\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin
...\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
...\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
...\.espressif\tools\cmake\3.16.4\bin
...\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin
...\.espressif\tools\ninja\1.10.0\
...\.espressif\tools\idf-exe\1.0.1\
...\.espressif\tools\ccache\3.7\
...\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
...\.espressif\python_env\idf4.2_py3.9_env\Scripts
...\embedded_workspace\freertos-release\vendors\espressif\esp-idf\tools
%PATH%
Checking if Python packages are up to date...
Python requirements from ...\embedded_workspace\freertos-release\vendors\espressif\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
    idf.py build

PS ...\embedded_workspace\freertos-release\vendors\espressif\esp-idf> cd ../../../
PS ...\embedded_workspace\freertos-release> idf.py build
& : The term '\python.exe' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct and
try again.
At ...\.espressif\Initialize-Idf.ps1:26 char:20
+ function idf.py { &$PythonCommand "$IDF_PATH\tools\idf.py" $args }
+                    ~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (\python.exe:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
```

georgik, 2021-07-16:
@laszlopataki Installer 2.3 is deprecated, please use latest version (e.g. 2.9) from: https://dl.espressif.com/dl/esp-idf/
You can download version also from releases.

Please, just run the installer, pick existing ESP-IDF installation on your hard-drive and installer will repair missing files.

BTW: It should not be even possible to download 2.3 installer. The link on server should point to latest stable installer 2.9. Can you give me a hint where you found the binary?

georgik, 2021-07-16:
Please, correct me if I'm wrong: You'd like to start the shell in different location that root directory of ESP-IDF.

This is not possible in current version, because the Working directory is used to determine identification of ESP-IDF.

It should be possible to achieve the behavior and remove dependency on Working directory by changing implementation of the initializer a little and use IDF ID in the link target specification. I'll validate the idea.

laszlopataki, 2021-07-16:
Hi @georgik , 
Thank you for the prompt reply. 

I got an email about your question. Let me reply so you might have the chance to sync with the other teams.  
Your question was: 
_BTW: It should not be even possible to download 2.3 installer. The link on server should point to latest stable installer 2.9. Can you give me a hint where you found the binary?_
The answer is: 
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#setup-espressif-idf42
Please follow the link to windows installer 
https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/get-started/windows-setup.html#esp-idf-tools-installer

> @laszlopataki This is limitation. Current version of deployment does not allow to relocate ESP-IDF after the installation.
> 
> This is caused by fact that ID of ESP-IDF is computed from path. It should be path independent so that it's possible to relocate ESP-IDF.
> 
> You're right, in case of missing value the tool should stop and inform user what went wrong and how it could be corrected.
> 
> The simple way how to fix relocated ESP-IDF links is to run the installer again and pick the existing location of ESP-IDF.
> 
> Other option is to run idf-env and register the path. See more at https://github.com/espressif/idf-env#working-with-configuration

Yes, I reinstalled and it is fine as soon as you have one main project with AFR, but obviously, it is not a good solution. 
Thank you to bring my attention to [this](https://github.com/espressif/idf-env#working-with-configuration) it is a good workaround. 👍 

> Please, correct me if I'm wrong: You'd like to start the shell in different location that root directory of ESP-IDF.
>
> This is not possible in current version, because the Working directory is used [...]

Yes, what I tried is option 1 in [this](https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/get-started/windows-setup.html#using-the-command-prompt) espressif guide. To be specific I change the "Start in" field of the shortcut to the esp-idf root directory. _C:\example\freertos\vendors\espressif\esp-idf_

georgik, 2021-07-16:
@laszlopataki Thank you for confirmation. I've scheduled the feature for release 2.10

georgik, 2021-07-29:
Available in Releases: idf-installer 2.9 beta 04.

PR #34 (2021-07-26): tools: support for launching ESP-IDF from different directory
Add support for launching ESP-IDF from different working directory.
It's based on esp-idf ID which is generated during installation. Instead of referencing by PATH it's referenced by the ID.
This allows to specify Working directory in LNK files.
The update contains correction for PowerShell, CMD and Windows Terminal Fragments.
The solution should be backward compatible.

Closes #31 #29
commit 23d59052 (2021-07-26): tools: support for launching ESP-IDF from different directory (#34)

Closes #31 #29
closed 2021-07-29 by georgik