espressif/idf-installer #39: McAfee antivirus blocks installation for long period of time

JurajSadel, 2021-08-02:
How to simulate: Start installer, click install and it will block the installation

georgik, 2021-08-02:
Identification of antivirus diagnosed by idf-env: `McAfee VirusScan`

Problem: Significant blocking delay when launching idf-env for driver installation in elevated privileges mode - 10+ seconds

Workaround: In component selection uncheck `Drivers - Requires elevation of privileges`. The installation will pass without waiting. Drivers could be installed later on by command:
```
idf-env.exe driver install --espressif --ftdi --silabs
```

Note: Turning off `Real-time scanning` does not have impact on the delay

Possible solution: Launch idf-env in background

Requested change: Inform user on diagnostic screen about potential problem with this particular antivirus.

georgik, 2021-09-17:
Routing invoking idf-env is blocking. It should be changed to async mode.

georgik, 2021-09-24:
Solved in release in 2.11

closed 2021-09-24 by georgik