espressif/idf-installer #50: Online Installer Hang on Windows 7 sp1

dsbonomi, 2021-08-28:
In trying to correct a problem with JTAG, I removed and started to re-install the ESP-IDF environment.
I know Win7 is no longer supported by Microsoft but I have a few tools that will not run on Win10. 

I chose to try the online installer. I was less than completely sucessful. 

Where the installer window hung. 

[image]

The Full Log
[image]

Running the certificates script outside of the installer

[image]

Installer Window after clicking stop 

[image]

The tool was able to continue and get a 624 MByte download for the ESP-IDF 

Powershell test failed at finish:

[image]
 
Command (Cmd32)  succeded.

[image]

Please note, the instructions seem to imply loading the ESP-IDF first. Then adresses vs_code. The VS_Code installer for the ESP-IDF Extension does not support already having an esp-idf directory, and forces loading a second copy.

georgik, 2021-08-28:
@dsbonomi Thank you for reporting the problem. 
Old Windows 7 with PowerShell before 2.0 might stop on this error.

There are 3 possible workarounds:
- download Offline installer
- upgrade PowerShell
- open in IE url https://dl.espressif.com to update cache of root certificates which is used by Python, that's basically what the ps1 script is doing. Click Stop button and Next on System diagnostic check to skip this check.

We will prepare correction without PowerSell in next version.

georgik, 2021-08-31:
@dsbonomi BTW: If you need just to correct problem with JTAG, you can use idf-env to deploy new driver.
```
Invoke-WebRequest 'https://dl.espressif.com/dl/idf-env/idf-env.exe' -OutFile .\idf-env.exe
.\idf-env.exe driver install --espressif
```
If your PowerShell does not support Invoke-WebRequest, simply download idf-env from: https://github.com/espressif/idf-env/releases

georgik, 2021-08-31:
@dsbonomi Please, can you make a test on your Windows 7 of updated installer? 
https://github.com/espressif/idf-installer/releases/tag/online-2.11-beta-01
The Certificate check should pass without problem.

dsbonomi, 2021-08-31:
I have downloaded and executed the referenced software, It did not hang.  It ran through the certificate check with no problem,  There does seem to be a small issue with the Powershell initialization script as shown below.  Python is there, but the 

[image]

Python is where it is supposed to be.

[image]

The named python file is present in the zip file I downloaded. It does not seem to be present anywhere else.

georgik, 2021-09-01:
@dsbonomi Thank you for the test and confirmation.

The Python issue in PowerShell seems to be caused by missing IDF_PATH which is not correctly set by the PowerShell script and then the script is trying to execute /tools/idf_tools.py, but the prefix path to that tools is missing. Let me investigate the PowerShell script and check compatibility with older Windows.

georgik, 2021-09-01:
@dsbonomi the problem seems to be somwehre in `.espressif/Initialize-Idf.ps1`.

Please open CMD where ESP-IDF works and type following command:
```
idf-env config get
```
This should give a list installed ESP-IDFs.

Then enter following command and replace IDF_ID with one of installed ESP-IDF ids displayed by previous command which has format `esp-idf-md5hash`:
```
idf-env  get --property path --idf-id IDF_ID
```
This should give the path to ESP-IDF. The same ID should be also part of Desktop Lnk to PowerShell (right-click properties on lnk icon). Please, post here also what you can see in Desktop Lnk to PowerShell in the field: Target.
There should be something like this:
```
C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -ExecutionPolicy Bypass -NoExit -File "C:\Users\user\.espressif/Initialize-Idf.ps1" -IdfId esp-idf-22d12587f01f0528f6edf64c7ba5f785
```

dsbonomi, 2021-09-05:
Both the CMD and PowerShell are stating:
Setting PYTHONNOUSERSITE, was not set
at initialization. CMD Runs to Ready to compile.
PS Bails at not finding idf-tools.py  

At this time, I do not have an explanation for the change. 

**First Command Request:**
C:\Users\Scott\Desktop\esp-idf>idf-env config get
{
    "$schema": "http://json-schema.org/schema#",
    "$id": "http://dl.espressif.com/dl/schemas/esp_idf",
    "_comment": "Configuration file for ESP-IDF Eclipse plugin.",
    "_warning": "Use / or \\ when specifying path. Single backslash is not allow
ed by JSON format.",
    "gitPath": "C:/Users/Scott/.espressif/tools/idf-git/2.30.1/cmd/git.exe",
    "idfToolsPath": "C:/Users/Scott/.espressif",
    "idfSelectedId": "esp-idf-234602bf712e6e817861a94c77ff87e9",
    "idfInstalled": {
        "esp-idf-234602bf712e6e817861a94c77ff87e9": {
            "version": "4.3",
            "python": "C:/Users/Scott/.espressif/python_env/idf4.3_py3.8_env/Scr
ipts/python.exe",
            "path": "C:/Users/Scott/Desktop/esp-idf/"
        },
        "esp-idf-f3ebd569a6d997fbe30e90360a663062": {
            "version": "4.3",
            "python": "C:\\Users\\Scott\\.espressif\\python_env\\idf4.3_py3.8_en
v\\Scripts\\python.exe",
            "path": "c:\\Users\\Scott\\Desktop\\esp\\esp-idf"
        }
    }
}

**Second Command Request**
From Cmd:
C:\Users\Scott\Desktop\esp-idf> idf-env get--property--path  --idf-id 234602b
f712e6e817861a94c77ff87e9
error: error: Found argument 'get--property--path' which wasn't expected, or isn
't valid in this context
From PowerShell: C:\Users\Scott\Desktop\esp-idf> idf-env get--property--path  --idf-id 234602b
f712e6e817861a94c77ff87e9
error: error: Found argument 'get--property--path' which wasn't expected, or isn
't valid in this context

Some experimentation, and poor hints from the help option, got me to: 
C:\Users\Scott\Desktop\esp-idf>   idf-env config  get -i 234602bf712e6e817861a94c77ff87e9
{
    "$schema": "http://json-schema.org/schema#",
    "$id": "http://dl.espressif.com/dl/schemas/esp_idf",
    "_comment": "Configuration file for ESP-IDF Eclipse plugin.",
    "_warning": "Use / or \\ when specifying path. Single backslash is not allowed by JSON format.",
    "gitPath": "C:/Users/Scott/.espressif/tools/idf-git/2.30.1/cmd/git.exe",
    "idfToolsPath": "C:/Users/Scott/.espressif",
    "idfSelectedId": "esp-idf-234602bf712e6e817861a94c77ff87e9",
    "idfInstalled": {
        "esp-idf-234602bf712e6e817861a94c77ff87e9": {
            "version": "4.3",
            "python": "C:/Users/Scott/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe",
            "path": "C:/Users/Scott/Desktop/esp-idf/"
        },
        "esp-idf-f3ebd569a6d997fbe30e90360a663062": {
            "version": "4.3",
            "python": "C:\\Users\\Scott\\.espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe",
            "path": "c:\\Users\\Scott\\Desktop\\esp\\esp-idf"
        }
    }
}

**Request Three** 
**Target Field for ESP-IDF 4.3 PowerShell Property Dialog.:**
C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -ExecutionPolicy Bypass -NoExit -File "C:\Users\Scott\.espressif/Initialize-Idf.ps1" -IdfId esp-idf-234602bf712e6e817861a94c77ff87e9

georgik, 2021-09-06:
@dsbonomi Thank you very much for the report.
The problem is caused by export.ps1:
```
$IDF_PATH = $PSScriptRoot
```
PSScriptRoot is valid only in PowerShell 3. https://riptutorial.com/powershell/example/27231/-psscriptroot

To workaround this problem it's sufficient to comment out line where IDF_PATH is set from script path in export.ps1.

The problem is that rest of the script won't be evaluated under PowerShell 2.
Error:
```
Combine : Cannot find an overload for "Combine" and the argument count: "4".
```
The minimal requirement to run the script is PowerShell 3 which was released in 2012. The requirement is not set anywhere.

To simulate the problem in newer PowerShell it's sufficient to append option "-version 2" to command line with powershell.exe.

georgik, 2021-09-06:
@dsbonomi Please, could you try to install PowerShell 3 on your Windows 7 SP1?
https://www.microsoft.com/en-us/download/details.aspx?id=34595

georgik, 2021-09-24:
Solved in release in 2.11

closed 2021-09-24 by georgik