espressif/idf-installer #54: Installation failed with Non-zero exit status 1.

ua100k, 2021-09-06:
I have uninstalled the previous installations, deleted all the files, cleaned all  directories but keep getting the same error. 

File "subprocess.py", line 364, in check_call
                          subprocess.CalledProcessError: Command '['C:\\esp\\tools\\.espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe', '-m', 'pip', 'install', '--no-warn-script-location', '-r', 'C:\\esp\\esp-idf\\requirements.txt', '--extra-index-url', 'https://dl.espressif.com/pypi', '--find-links', 'C:\\esp\\tools\\.espressif\\tools\\idf-python-wheels\\idf4.3_py3.9_2021-01-07']' returned non-zero exit status 1.

Not sure what to do.

georgik, 2021-09-06:
@ua100k Thank you for reporting the problem. Which installer do you use? 2.10 Online?
Please delete directory: `C:\esp\tools\.espressif\python_env` and retry the installation.

ua100k, 2021-09-06:
yes I have deleted the whole directory under esp and everything else. So there is no esp32 files left.
I also did a system restore. I am using 2.10 online.  I am not able to get past this error.

georgik, 2021-09-06:
@ua100k Thank you for providing more details. 

Please submit also the log with error message. You can find it in AppData\Temp. It should be also displayed at the end of failed installation. Or you can locate it by click on "Full Log" button in the second screen of the installer with on 2nd installer screen - System Check.

ua100k, 2021-09-06:
[Setup Log 2021-09-06 #003.txt](https://github.com/espressif/idf-installer/files/7116775/Setup.Log.2021-09-06.003.txt)

Thank you for looking at this. Here is the log. 
Please let me know if you need anything else.

georgik, 2021-09-06:
Thank you @ua100k for the log. Whole installation seems to pass without problem. Location of Python and creation of virtualenv seems to be also correct. Please, can you try Offline installer which contains also IDF 4.3? 
https://github.com/espressif/idf-installer/releases/tag/offline-2.11-beta-02

I'll do further analysis of Python traceback.

georgik, 2021-09-06:
@ua100k The strange thing is this line:
```
seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=C:\Users\deutrium-tritium\AppData\Local\pypa\virtualenv
```
It seems that the Python is reusing some environment environment.
Please can you check that directory and remove it? It seems like a relict of some Python installation.

ua100k, 2021-09-06:
Uggg. I did 2 things at once so I don't know which one worked. I deleted the directory and used the offline release. 

Would you like me to try again with the offline or the online install?

georgik, 2021-09-06:
@ua100k Thank you for the confirmation. The real correction was in cleaning the AppData\Local\pypa.

This error happens sometimes when there was an installation of Python using User Site which stores packages in AppData\Local. The installer Python does not clear this directory even after uninstallation. Most of this cases are covered by setting environment variable PYTHONNOUSERUSERSITE
The question is why this settings didn't apply in your case and required manual correction.

It's not necessary to perform further tests. I have an idea how to simulate problematic behavior. Now you can use Online or Offline installer to install version of ESP-IDF that you like. Both installer supports installation of evn multiple ESP-IDFs. It's not necessary to run uninstall. If you need another installation just start the installer again and select different directory for the installation.

georgik, 2021-09-06:
@JurajSadel Please help us simulate the problem. Here is an idea how to achieve problematic state:
- install old Python 3.6 ideally something like 3.6.1. Choose model of installation to AppData directory
- start the command line and install virtualenv: `pip install virtualenv`
- check AppData\Local\pypa, it should exist
- uninstall Python 3.6, AppData\Local\pypa should be pressent
- now, start ESP-IDF installer and installation should fail on Python error, same to the log posted few lines above

JurajSadel, 2021-09-07:
It is possible to change seeder with option `--seeder pip`
- https://man.archlinux.org/man/virtualenv.1.en#Seeders
- https://virtualenv.pypa.io/en/latest/cli_interface.html#section-seeder

```
PS C:\Users\JurajSadel> $env:PYTHONNOUSERSITE="True"
PS C:\Users\JurajSadel> C:\Users\JurajSadel\.espressif\tools\idf-python\3.8.7\python.exe -m virtualenv "C:\Users\JurajSadel\.espressif\python_env\idf4.3_py3.8_env_test" -p "C:\Users\JurajSadel\.espressif\tools\idf-python\3.8.7\python.exe" --seeder pip
created virtual environment CPython3.8.7.final.0-64 in 3312ms
  creator CPython3Windows(dest=C:\Users\JurajSadel\.espressif\python_env\idf4.3_py3.8_env_test, clear=False, no_vcs_ignore=False, global=False)
  seeder PipInvoke(download=False, pip=bundle, setuptools=bundle, wheel=bundle)
    added seed packages: pip==20.3.3, setuptools==51.3.3, wheel==0.36.2
  activators BashActivator,BatchActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
PS C:\Users\JurajSadel> C:\Users\JurajSadel\.espressif\tools\idf-python\3.8.7\python.exe -m virtualenv "C:\Users\JurajSadel\.espressif\python_env\idf4.3_py3.8_env_test2" -p "C:\Users\JurajSadel\.espressif\tools\idf-python\3.8.7\python.exe"
created virtual environment CPython3.8.7.final.0-64 in 501ms
  creator CPython3Windows(dest=C:\Users\JurajSadel\.espressif\python_env\idf4.3_py3.8_env_test2, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=C:\Users\JurajSadel\AppData\Local\pypa\virtualenv)
    added seed packages: pip==20.3.3, setuptools==51.3.3, wheel==0.36.2
  activators BashActivator,BatchActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
  ```

georgik, 2021-09-07:
@ua100k The new build of installer with the correction is available: 
https://github.com/espressif/idf-installer/releases/tag/online-2.11-beta-03

closed 2021-09-07 by georgik

ua100k, 2021-09-07:
Many thanks for the very quick response and fix. The support is fantastic and Georgik actually took the time to look at the logs. Brilliant. And a fix issued. Many kudos to all.

PR #56 (2021-09-07): Feature/python seeder
Use explicit seeder. 
Closes #54