espressif/idf-installer #68: Installation has failed with exit code 1

Jan-Petter1, 2021-09-30:
Hi
Trying to install esp_idf_tools the installer report this error:
Installation has failed with exit code 1
Jan-Petter Bakke

georgik, 2021-09-30:
@Jan-Petter1 Thank you for reporting the issue. To diagnose the problem we need more information.

Please, use the latest installer https://github.com/espressif/idf-installer/releases/tag/online-2.12-beta-01
If the installation fails, please attach the installation log.
You can access it in your home directory under \AppData\Temp\Setup*.txt or simply by clicking Full log button on the second screen.

ingcicala, 2021-09-30:
I have same problem!!!

exitcode 1 : 
"
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1125)>
                          WARNING: Failed to download https://dl.espressif.com/dl/ninja-1.10.2-win64.zip to C:\ESP_Tool\.espressif\dist\ninja-1.10.2-win64.zip.tmp
                          Downloading ninja-1.10.2-win64.zip to C:\ESP_Tool\.espressif\dist\ninja-1.10.2-win64.zip.tmp
                          WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1125)>
                          WARNING: Failed to download https://dl.espressif.com/dl/ninja-1.10.2-win64.zip to C:\ESP_Tool\.espressif\dist\ninja-1.10.2-win64.zip.tmp
                          Downloading ninja-1.10.2-win64.zip to C:\ESP_Tool\.espressif\dist\ninja-1.10.2-win64.zip.tmp
                          WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1125)>
                          WARNING: Failed to download https://dl.espressif.com/dl/ninja-1.10.2-win64.zip to C:\ESP_Tool\.espressif\dist\ninja-1.10.2-win64.zip.tmp
                          ERROR: Failed to download, and retry count has expired
                          Traceback (most recent call last):
                            File "C:\ESP_IDF\\tools\idf_tools.py", line 1675, in <module>
                              main(sys.argv[1:])
                            File "C:\ESP_IDF\\tools\idf_tools.py", line 1671, in main
                              action_func(args)
                            File "C:\ESP_IDF\\tools\idf_tools.py", line 1322, in action_install
                              tool_obj.download(tool_version)
                            File "C:\ESP_IDF\\tools\idf_tools.py", line 705, in download
                              raise DownloadError()
                          __main__.DownloadError
"
[Setup Log 2021-09-30 #003.txt](https://github.com/espressif/idf-installer/files/7258131/Setup.Log.2021-09-30.003.txt)

ingcicala, 2021-09-30:
[SSL: CERTIFICATE_VERIFY_FAILED]  ?????

Jan-Petter1, 2021-09-30:
Still the same problem

2021-09-30 10:46:05.696   Log opened. (Time zone: UTC+02:00)
2021-09-30 10:46:05.696   Setup version: Inno Setup version 6.2.0
2021-09-30 10:46:05.696   Original Setup EXE: C:\Users\Jan-PetterBakke\Downloads\esp-idf-tools-setup-online-2.12-beta-01.exe
2021-09-30 10:46:05.696   Setup command line: /SL5="$E086C,2460577,831488,C:\Users\Jan-PetterBakke\Downloads\esp-idf-tools-setup-online-2.12-beta-01.exe" 
2021-09-30 10:46:05.696   Windows version: 10.0.19042  (NT platform: Yes)
2021-09-30 10:46:05.696   64-bit Windows: Yes
2021-09-30 10:46:05.696   Processor architecture: x64
2021-09-30 10:46:05.697   User privileges: None
2021-09-30 10:46:05.701   Administrative install mode: No
2021-09-30 10:46:05.701   Install mode root key: HKEY_CURRENT_USER
2021-09-30 10:46:05.701   64-bit install mode: Yes
2021-09-30 10:46:10.784   Created temporary directory: C:\Users\JAN-PE~1\AppData\Local\Temp\is-018CJ.tmp
2021-09-30 10:46:10.784   -- DLL function import --
2021-09-30 10:46:10.784   Function name: idpAddFile
2021-09-30 10:46:10.784   DLL name: files:idp.dll
2021-09-30 10:46:10.784   Extracting temporary file: C:\Users\JAN-PE~1\AppData\Local\Temp\is-018CJ.tmp\idp.dll
2021-09-30 10:46:10.804   Dest DLL name: C:\Users\JAN-PE~1\AppData\Local\Temp\is-018CJ.tmp\idp.dll
2021-09-30 10:46:10.804   Importing the DLL function.
2021-09-30 10:46:10.884   Successfully imported the DLL function. Delay loaded? No
2021-09-30 10:46:10.884   -- DLL function import --
2021-09-30 10:46:10.884   Function name: idpAddFileComp
2021-09-30 10:46:10.884   DLL name: files:idp.dll
2021-09-30 10:46:10.884   Dest DLL name: C:\Users\JAN-PE~1\AppData\Local\Temp\is-018CJ.tmp\idp.dll
2021-09-30 10:46:10.884   Importing the DLL function.
2021-09-30 10:46:10.884   Successfully imported the DLL function. Delay loaded? No
2021-09-30 10:46:10.884   -- DLL function import --
2021-09-30 10:46:10.884   Function name: idpAddMirror
2021-09-30 10:46:10.884   DLL name: files:idp.dll
2021-09-30 10:46:10.884   Dest DLL name: C:\Users\JAN-PE~1\AppData\Local\Temp\is-018CJ.tmp\idp.dll
2021-09-30 10:46:10.884   Importing the DLL function.
2021-09-30 10:46:10.884   Successfully imported the DLL function. Delay loaded? No
2021-09-30 10:46:10.884   -- DLL function import --
2021-09-30 10:46:10.884   Function name: idpAddFtpDir
2021-09-30 10:46:10.884   DLL name: files:idp.dll
2021-09-30 10:46:10.884   Dest DLL name: C:\Users\JAN-PE~1\AppData\Local\Temp\is-018CJ.tmp\idp.dll
2021-09-30 10:46:10.884   Importing the DLL function.
2021-09-30 10:46:10.884   Successfully imported the DLL function. Delay loaded? No
2021-09-30 10:46:10.884   -- DLL function import --
2021-09-30 10:46:10.884   Function name: idpAddFtpDirComp
2021-09-30 10:46:10.884   DLL name: files:idp.dll
2021-09-30 10:46:10.884   Dest DLL name: C:\Users\JAN-PE~1\AppData\Local\Temp\is-018CJ.tmp\idp.dll
2021-09-30 10:46:10.884   Importing the DLL function.
2021-09-30 10:46:10.884   Successfully imported the DLL function. Delay loaded? No
2021-09-30 10:46:10.884   -- DLL function import --
2021-09-30 10:46:10.884   Function name: idpClearFiles
2021-09-30 10:46:10.884   DLL name: files:idp.dll
2021-09-30 10:46:10.884   Dest DLL name: C:\Users\JAN-PE~1\AppData\Local\Temp\is-018CJ.tmp\idp.dll
2021-09-30 10:46:10.884   Importing the DLL function.
2021-09-30 10:46:10.884   Successfully imported the DLL function. Delay loaded? No
2021-09-30 10:46:10.884   -- DLL function import --
2021-09-30 10:46:10.884   Function name: idpFilesCount
2021-09-30 10:46:10.884   DLL name: files:idp.dll
2021-09-30 10:46:10.884   Dest DLL name: C:\Users\JAN-PE~1\AppData\Local\Temp\is-018CJ.tmp\idp.dll
2021-09-30 10:46:10.884   Importing the DLL function.
2021-09-30 10:46:10.884   Successfully imported the DLL function. Delay loaded? No
2021-09-30 10:46:10.884   -- DLL function import --
2021-09-30 10:46:10.884   Function name: idpFtpDirsCount
2021-09-30 10:46:10.884   DLL name: files:idp.dll
2021-09-30 10:46:10.884   Dest DLL name: C:\Users\JAN-PE~1\AppData\Local\Temp\is-018CJ.tmp\idp.dll
2021-09-30 10:46:10.884   Importing the DLL function.
2021-09-30 10:46:10.884   Successfully imported the DLL function. Delay loaded? No
2021-09-30 10:46:10.884   -- DLL function import --
2021-09-30 10:46:10.884   Function name: idpFileDownloaded
2021-09-30 10:46:10.884   DLL name: files:idp.dll
2021-09-30 10:46:10.884   Dest DLL name: C:\Users\JAN-PE~1\AppData\Local\Temp\is-018CJ.tmp\idp.dll
2021-09-30 10:46:10.884   Importing the DLL function.
2021-09-30 10:46:10.884   Successfully imported the DLL function. Delay loaded? No
2021-09-30 10:46:10.884   -- DLL function import --
[... 336 lines cut ...]
                          Skipping ***@***.*** (already installed)
                          Skipping ***@***.*** (already installed)
                          Skipping ***@***.*** (already installed)
                          Installing ***@***.***
                          Downloading ninja-1.10.2-win64.zip to C:\Users\Jan-PetterBakke\.espressif\dist\ninja-1.10.2-win64.zip.tmp
                          WARNING: Failed to download https://dl.espressif.com/dl/ninja-1.10.2-win64.zip to C:\Users\Jan-PetterBakke\.espressif\dist\ninja-1.10.2-win64.zip.tmp
                          Downloading ninja-1.10.2-win64.zip to C:\Users\Jan-PetterBakke\.espressif\dist\ninja-1.10.2-win64.zip.tmp
[... cut ...]
                            File "C:\Users\Jan-PetterBakke\Desktop\esp-idf\\tools\idf_tools.py", line 705, in download
                          
2021-09-30 10:46:46.878   --------
2021-09-30 10:46:46.878   Message box (OK):
2021-09-30 10:46:49.677   User chose OK.
2021-09-30 10:46:49.682   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2021-09-30 10:46:52.076   User chose Yes.
2021-09-30 10:46:52.114   CurStepChanged raised an exception.

georgik, 2021-09-30:
@Jan-Petter1 Please try verify whether you can download this URL directly in your web browser:
https://dl.espressif.com/dl/ninja-1.10.2-win64.zip

If yes, please retry the installation. The certificate cache should be refreshed by the download from browser.

ingcicala, 2021-09-30:
Offline installer tools works ....

Jan-Petter1, 2021-09-30:
No difference

https://dl.espressif.com/dl/ninja-1.10.2-win64.zip

ingcicala, 2021-09-30:
offline installer tool works but it doesn't install eclipse .....
I will retry online tool ...

ingcicala, 2021-09-30:
Idem
No difference ......

> @Jan-Petter1 Please try verify whether you can download this URL directly in your web browser: https://dl.espressif.com/dl/ninja-1.10.2-win64.zip
> 
> If yes, please retry the installation. The certificate cache should be refreshed by the download from browser.

georgik, 2021-09-30:
@ingcicala Thank you for verification. That's an interesting issue.

Please try previous version of installer 2.10 which contains PowerShell code which refreshes Windows certificates, the method was changed in 2.11.
https://github.com/espressif/idf-installer/releases/tag/offline-2.10

Please, attach the install log, so that we can diagnose the problem.

ingcicala, 2021-09-30:
> offline installer tool works but it doesn't install eclipse ..... I will retry online tool ...

[image]

ingcicala, 2021-09-30:
online version 2.10 has the same problem
offline version 2.11 works but at the end it doesn't install eclipse ....

finally offline versione 2.10 works !!!!!!!!!!!!!!!!!!!!!!!!!

ingcicala, 2021-09-30:
eclipse in offline 2.10 doen't recognise IDF_PATH ..... uff

Jan-Petter1, 2021-09-30:
Thanks, it worked

https://github.com/espressif/idf-installer/releases/tag/offline-2.10

georgik, 2021-09-30:
@ingcicala Thank you for posting details about 2.11 Offline problem with Eclipse deployment. I've created new issue to track this problem #69 

@Jan-Petter1 Thank you for confirmation that solution worked for you.

@ingcicala Regarding SSL: CERTIFICATE_VERIFY_FAILED on your Windows. Do you have some Windows Group Policy restrictions installed (often case in corporate notebooks) or different antivirus than Windows Defender?

I will also double check Python cryptography which seems to cause some troubles in the version released yesterday.

georgik, 2021-10-01:
If anybody else experience this issue, please add it to the comment. Please, provide Windows version (command winver), Windows Edition (home, enterprise, pro) and type of antivirus software (e.g. Windows Defender).

georgik, 2021-10-04:
The problem seems to be caused by outdated root certificates in the system: https://techcrunch.com/2021/09/21/lets-encrypt-root-expiry/?guccounter=1

georgik, 2021-10-19:
File server dl.espressif.com was moved to cloud storage with different certificate provider.
Closing the issue.

Feel free to reopen. In case of reopen please attach the log from the installer.

closed 2021-10-19 by georgik

xiaolongba, 2021-11-01:
@georgik it seems that I also encountered this issue, this is not the specific case, it should be the same as others, the details as below:
[image]

the only workaround is to return back to https://github.com/espressif/idf-installer/releases/tag/offline-2.10 and then install the tools and then you can use the latest version https://github.com/espressif/idf-installer/releases/tag/online-2.12-beta-07 at this moment.