espressif/idf-installer #72: flasher_args.json file is missing from the build directory, can't proceed, please build properly!!

ua100k, 2021-10-08:
Hi, I have done the install of Visual Studio Code using the video provided espressif  - https://www.youtube.com/watch?v=Lc6ausiKvQM

One issue was that the scripts could not be executed in powershell due to group policy. I fixed that. 

I can build the code but I cannot flash it. This is the error that I get. 

I see this file in the esp-idf sub directory but shouldn't the build generate this file? 
Also where do I put this file so that I can flash using Visual Studio. 

Thank you.

georgik, 2021-10-11:
@ua100k Thanks for reporting the issue. I believe, that the issue belongs to: https://github.com/espressif/vscode-esp-idf-extension

@brianignacio5 Do you have any quick hint?

brianignacio5, 2021-10-11:
Thanks for reporting this. I'll move this to the vscode-extension repository.