espressif/idf-installer #83: 【Bug Report】the option for espressif download instead of github can not be chose

xiaolongba, 2021-10-30:
Hi Espressif experts,

the following option cannot be chose when installing the idf-installer, the details as below:

[image]

could you give me a hand?

thanks in advance.

georgik, 2021-11-01:
@xiaolongba Thank you for reporting the issue. This is really the bug in the installer.
The bug seems to be in all installers since version 2.6.
You can try old installer 2.5 - https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
We will prepare fix for the issue.

georgik, 2021-11-01:
@xiaolongba Please test the new version of installer: https://github.com/espressif/idf-installer/releases/tag/online-2.12-beta-07

Basically the feature just sets environment variable `IDF_GITHUB_ASSETS` to value `dl.espressif.com/github_assets`

xiaolongba, 2021-11-01:
> Basically the feature just sets environment variable `IDF_GITHUB_ASSETS` to value `dl.espressif.com/github_assets`

Can I configure it in the environment variable manually? if yes, which one?
[image]

> Please test the new version of installer: https://github.com/espressif/idf-installer/releases/tag/online-2.12-beta-07

I only tried to use the offline version, but the result is the same.

xiaolongba, 2021-11-01:
can you release the latest offline version for testing only, I can try and feedback to you soon.

georgik, 2021-11-01:
@xiaolongba The checkbox make sense only for Online installer.

The limitation of Inno Setup is that it includes the check box also in Offline installer. Hm, you gave me an idea, that we can add this feature also to Offline installer where the result will be configured environment variable.

Regarding the variable: You can set the variable in `User variables` and start new Terminal/CMD so that the variable is loaded after changing the variable.

xiaolongba, 2021-11-01:
> The limitation of Inno Setup is that it includes the check box also in Offline installer. Hm, you gave me an idea, that we can add this feature also to Offline installer where the result will be configured environment variable.

glad to see that. as you know, we hope to download the tools from dl.espressif.com/github_assets as the limitation for accessing GitHub in China. leads to the speed for downloading being very very slow.

xiaolongba, 2021-11-01:
one more question, the speed for downloading can be improved if setting the environment variable IDF_GITHUB_ASSETS to value dl.espressif.com/github_assets when the limitation for accessing Github?

georgik, 2021-11-01:
Recently we started to mirror our assets at AWS S3 which should be reachable from China at full speed.
Links look like this: https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.11.exe

Yes, setting this env variable should influence all the tooling which is invoking install.bat.

xiaolongba, 2021-11-01:
> Recently we started to mirror our assets at AWS S3 which should be reachable from China at full speed.
> Links look like this: https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.11.exe

got it, since the 2.11 version, all resources can be reached from China at full speed. is it correct?

> Yes, setting this env variable should influence all the tooling which is invoking install.bat.

it is the same as above, all resources can be reached from China at full speed after setting this env variable?

xiaolongba, 2021-11-01:
> Please test the new version of installer: https://github.com/espressif/idf-installer/releases/tag/online-2.12-beta-07
> 
> Basically the feature just sets environment variable `IDF_GITHUB_ASSETS` to value `dl.espressif.com/github_assets`

sorry, Bro. I gave you the wrong message. actually, I used the online version for tools installing. it works well with using this version and the downloading at full speed after enable this checkbox.

however, I tried to add the  environment variable `IDF_GITHUB_ASSETS` to value `dl.espressif.com/github_assets, nothing happen.

xiaolongba, 2021-11-04:
Hi @georgik Bro,

can you help me with the doubts I mentioned above? I promise that these are the last questions😂

georgik, 2021-11-22:
Thank you for the questions @xiaolongba .
If you set the variable in Windows Environment variable, please make sure to open new terminal after adding the value. Environment variables are loaded from the system only when process starts. Other option in Powershell is to set it in runtime (it will be gone in next start): `$env:IDF_GITHUB_ASSETS="dl.espressif.com/github_assets"`.
Since the solution works, I'm closing this issue.
I've created new one for adding a support to the installer to persist the variable: #93

closed 2021-11-22 by georgik

georgik, 2021-11-22:
Note: When the variable is set the download message should look like:
```
Using GitHub assets mirror for URL: https://github.com/espressif/crosstool-NG/releases/download/esp-2021r1/xtensa-esp32-elf-gcc8_4_0-esp-2021r1-win64.zip => https://dl.espressif.com/github_assets/espressif/crosstool-NG/releases/download/esp-2021r1/xtensa-esp32-elf-gcc8_4_0-esp-2021r1-win64.zip
```

xiaolongba, 2021-11-22:
thank you, bro

PR #90 (2021-11-19): bugfix: enable selection of download mirror
Closes #83