espressif/idf-installer #87: Doesn't Support Windows Arm architecture

drony, 2021-11-06:
I think just installer restriction. so many x86 and x64 apps works normally.
I tried Mac OS with parallels and windows 11 insider

error message attached
[image]

drony, 2021-11-06:
ok i fixed 
i deleted to lines from idfToolsSetup.iss and compiled from source code. works very well

ArchitecturesAllowed=x64
ArchitecturesInstallIn64BitMode=x64

georgik, 2021-11-08:
Hi @drony . Thank you for the report.

The installer is intended to work only on Windows x64. Removing those lines might allow the installation on Windows 32-bit which is not supported by several tools inside ESP-IDF.

BTW: Is there a reason not to use native ESP-IDF for macOS and run it in Parallels? It should work pretty well with M1. I'm curious about the use-case.

georgik, 2021-11-22:
Closing this issue since ESP-IDF is natively supported on macOS aarch64 and macOS x86_64. It's also supported in Podman/Docker container.

Feel free to reopen the issue and add reason why it might make sems to run ESP-IDF in emulated environment like Parallels.

closed 2021-11-22 by georgik

cmsd2, 2024-08-12:
thanks. i used this workaround to get the installer to run on Windows-on-Arm on a snapdragon x elite