espressif/idf-installer #100: Installation has failed with exit code 101

chrissMD, 2021-12-15:
After trying to install ESP-IDF Windows Installer Download: [from here](https://dl.espressif.com/dl/esp-idf/?idf=4.4) I get the above error. The same happens for online and offline installation. The process does not continue further and no folder is created (using an online installation wizard).
When trying offline installation a folder is created, however, eventually, I receive the same error at the end of the installation wizard.[image]
[image]

georgik, 2021-12-15:
@chrissMD Please can you post content of System Check screen which is the second screen?
This looks like problems with Antivirus.

Please attach also the install log, which should open after failed installation. You can find it in `C:\Users\USERNAME\AppData\Local\Temp\Setup....txt`

georgik, 2021-12-15:
Potential workarounds: Uncheck driver installation in previous screen. The installation should proceed.
Install drivers manually via idf-env:
```
Invoke-WebRequest 'https://dl.espressif.com/dl/idf-env/idf-env.exe' -OutFile .\idf-env.exe; .\idf-env.exe driver install --espressif --ftdi --silabs
```
Or deploy them manually: https://github.com/espressif/idf-installer#manual-installation-of-drivers

chrissMD, 2021-12-20:
> Please attach also the install log, which should open after failed installation. You can find it in `C:\Users\USERNAME\AppData\Local\Temp\Setup....txt`
[Setup Log 2021-12-20 #001.txt](https://github.com/espressif/idf-installer/files/7744157/Setup.Log.2021-12-20.001.txt)

chrissMD, 2021-12-20:
> Potential workarounds: Uncheck driver installation in previous screen. The installation should proceed. Install drivers manually via idf-env:
>
> ```
> Invoke-WebRequest [...]

Can you be more specific please on how I can deploy necessary drivers manually? I am a total beginner :/

chrissMD, 2021-12-20:
> Potential workarounds: Uncheck driver installation in the previous screen. The installation should proceed. Install drivers manually via idf-env:
> 
> ```
> Invoke-WebRequest 'https://dl.espressif.com/dl/idf-env/idf-env.exe' -OutFile .\idf-env.exe; .\idf-env.exe driver install --espressif --ftdi --silabs
> ```
> 
> Or deploy them manually: https://github.com/espressif/idf-installer#manual-installation-of-drivers

I followed this advice. After unchecking the driver's installation the process went smoothly. After that, I manually installed the drivers following the instructions as described in the "Manual Installation of drivers" guide: [here](https://github.com/espressif/idf-installer#manual-installation-of-drivers). I could find the installed drivers under `C:\Users\user\.espressif\tools\idf-driver` three drivers, namely:

> ftdi-2021-05-03
> idf-driver-esp32-usb-jtag-2021-07-15
> silabs-2021-05-03

Look fine now, OR?

[image]

georgik, 2021-12-20:
@chrissMD Thank you for detailed report.
The installation looks correct. Drivers should be in place.
Now you can launch the installer, just unchek the option with drivers in Component selection screen, since drivers are already installed.

I'm still wondering what caused the initial crash on your computer. Please could you check your `AppData\Local\Temp\` for the log file which ends with 101 error? (Setup Log-2021...txt)

chrissMD, 2021-12-20:
> @chrissMD Thank you for detailed report. The installation looks correct. Drivers should be in place.
> 
> I'm still wondering what caused the initial crash on your computer. Please could you check your `AppData\Local\Temp\` for the log file which ends with 101 error? (Setup Log-2021...txt)
[check the above comment for the log file](https://github.com/espressif/idf-installer/issues/100#issuecomment-997751645)

chrissMD, 2021-12-20:
I run the installer with administrator rights, that is the best solution i think! That was the best solution so far. BR

georgik, 2021-12-21:
Thank you @chrissMD for finding the workaround in launching the installer in Administrator mode.

The problem seems to be related to elevation of privileges process.
When you run the installer in elevated mode, e.g. as Administrator, it does not require idf-env to switch to elevated mode.
So the failure seems to be somewhere in the process of elevating privileges.

Normally installer should not require to be executed in Administrator mode. Some steps might request elevation of privileges. E.g. installation of drivers or registration of Windows Defender rules.

I reopen the issue to make additional investigation. User should be informed about problem and solution. Crash message with code is not the best option from UX perspective.

georgik, 2021-12-21:
The problem might occur in older installer when one of sites which is hosting drivers is not reachable.
Fixed in https://github.com/espressif/idf-installer/releases/tag/online-2.13-beta-07
The installer will log warning about failed download and the installation process will continue.
The it's possible to retry the installation of drivers by `idf-env driver install --ftdi --silabs --espressif`

closed 2021-12-21 by georgik

suneelmb2410, 2024-09-02:
First step: While installing uncheck drivers.....
then download drivers zip file seperately using the below link(The response time is very slow)
https://www.wch.cn/downloads/file/314.html
the rename it as "whc-ch343ser.zip" and replace it in "Espressif\dist"  power shell enter the below command hit enter.......
This wroked for me.......