espressif/idf-installer #109: ESP-IDF Windows Installation fails

urikan1, 2022-01-16:
What could be the cause?

Traceback (most recent call last):
  File "C:\Users\user\esp-idf\\tools\idf_tools.py", line 1675, in <module>
    main(sys.argv[1:])
  File "C:\Users\user\esp-idf\\tools\idf_tools.py", line 1671, in main
    action_func(args)
  File "C:\Users\user\esp-idf\\tools\idf_tools.py", line 1323, in action_install
    tool_obj.install(tool_version)
  File "C:\Users\user\esp-idf\\tools\idf_tools.py", line 723, in install
    strip_container_dirs(dest_dir, self._current_options.strip_container_dirs)
  File "C:\Users\user\esp-idf\\tools\idf_tools.py", line 389, in strip_container_dirs
    rename_with_retry(path, tmp_path)
  File "C:\Users\user\esp-idf\\tools\idf_tools.py", line 375, in rename_with_retry
    os.rename(path_from, path_to)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\user\\.espressif\\tools\\cmake\\3.16.4' -> 'C:\\Users\\user\\.espressif\\tools\\cmake\\3.16.4.tmp'

georgik, 2022-01-24:
Thank you @urikan1 

The problem might be caused by antivirus installed on the system. 
Which antivirus is installed on the system?
You can check also the second screen of installer.

Some antiviruses allows to disable scanning for small period of time.

The problem is caused by renaming cmake package from cmake.tmp directory. The renaming fails, because the antivirus is scanning the directory. This was noticed in the past with Comodo antivirus.

georgik, 2022-02-01:
@urikan1 Please try one of Offline installers. They do not move cmake, so the problem should not occur.
https://github.com/espressif/idf-installer#single-esp-idf-branch-offline-installer

closed 2022-02-01 by georgik