espressif/idf-installer #110: Support constraint files in the offline installer

dobairoland, 2022-01-17:
ESP-IDF v5.0 will soon start to use a constraint file for installing Python packages (https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.0.txt). The offline installer would probably want to bundle this file and install the Python environment with care so it would remain "offline".

The user should be asked which desired features should be supported. `core` is mandatory, and currently there is only one optional one: `gdbgui`.  We might introduce more in the near future.

The online installer is pretty straightforward. Run `install.bat` for just the core features, or run `install.bat --enable-gdbgui` for supporting `gdbgui` as well. The corresponding `idf_tools.py` commands are `idf.py install-python-env` and `idf.py install-python-env --features=core,gdbgui`. 

The process could be split up in order to implement offline installing.

1. Put `espidf.constraints.vX.Y.txt` into `~/.espressif` where `vX.Y` is the ESP-IDF version and `X >= 5`.
2. Install the empty virtual Python environment.
3. Run `<virtualenv_python> -m pip install -c ~/.espressif/espidf.constraints.vX.Y.txt -r $IDF_PATH/requirements.core.txt -r $IDF_PATH/requirements.gdbgui.txt`. Omit `requirements.gdbgui.txt` if the feature is not desired.  
3. Run `$IDF_TOOLS/tools/idf_tools.py install-python-env --no-constraints --features=core,gdbgui` to update `idf-env.json` with the features. Alternatively, the file can be edited manually without running this command. Omit `gdbgui` if that feature is not required.

Please excuse my Unix-style commands. Please let me know if I can make this more comfortable for you on the ESP-IDF side or can provide more information. Thanks!

dobairoland, 2022-01-17:
PTAL @georgik @JurajSadel

georgik, 2022-02-01:
Update of existing ESP-IDF which is cloned from master is not working. The installer blocks installation on ESP-IDF directory selection. Thnx @mfialaf for finding the issue.

dobairoland, 2022-03-02:
There has been some recent changes related to this: https://github.com/espressif/esp-idf/commit/fddc73759ebb6f1b00306671474fca7ef90784be.

georgik, 2022-12-06:
IMO The repo with constraint file shoult be published as git repo. Handling this extra dependency by downloading it from https location is not very systematic and fragile. Also including version number in file name make things unecessary complex. Consider using other solution which has simpler maintenance in the future.

PR #111 (2022-02-01): offline: single branch installation and Espressif-IDE
Closes #92 
Closes #85 
Closes #110

Changing Offline Installer concept to Single Branch installer.
Changed Eclipse IDE to Espressif-IDE.
PR #155 (2022-12-06): add offline version 5.0
Closes #110 
Closes #150
closed 2022-12-06 by georgik