espressif/idf-installer #112: Installation has failed with exit code 1

dasom96, 2022-01-28:
Please help me install (Windows)
https://dl.espressif.com/dl/esp-idf/?idf=4.4 (offline)

2022-01-28 13:29:39.904   Time stamp of our file: 2021-11-22 16:30:56.000
2022-01-28 13:29:39.904   Installing the file.
2022-01-28 13:29:39.904   Creating directory: C:\eps\.espressif\tools\idf-driver\silabs-2021-05-03\arm
2022-01-28 13:29:39.910   Successfully installed the file.
2022-01-28 13:29:39.910   -- File entry --
2022-01-28 13:29:39.912   Dest filename: C:\eps\.espressif\tools\idf-driver\silabs-2021-05-03\arm64\silabser.sys
2022-01-28 13:29:39.912   Time stamp of our file: 2021-11-22 16:30:56.000
2022-01-28 13:29:39.912   Installing the file.
2022-01-28 13:29:39.912   Creating directory: C:\eps\.espressif\tools\idf-driver\silabs-2021-05-03\arm64
2022-01-28 13:29:39.919   Successfully installed the file.
2022-01-28 13:29:39.919   -- File entry --
2022-01-28 13:29:39.921   Dest filename: C:\eps\.espressif\tools\idf-driver\silabs-2021-05-03\x64\silabser.sys
2022-01-28 13:29:39.921   Time stamp of our file: 2021-11-22 16:30:56.000
2022-01-28 13:29:39.921   Installing the file.
2022-01-28 13:29:39.921   Creating directory: C:\eps\.espressif\tools\idf-driver\silabs-2021-05-03\x64
2022-01-28 13:29:39.928   Successfully installed the file.
2022-01-28 13:29:39.928   -- File entry --
2022-01-28 13:29:39.930   Dest filename: C:\eps\.espressif\tools\idf-driver\silabs-2021-05-03\x86\silabser.sys
2022-01-28 13:29:39.930   Time stamp of our file: 2021-11-22 16:30:56.000
2022-01-28 13:29:39.930   Installing the file.
2022-01-28 13:29:39.930   Creating directory: C:\eps\.espressif\tools\idf-driver\silabs-2021-05-03\x86
2022-01-28 13:29:39.937   Successfully installed the file.
2022-01-28 13:29:39.940   -- Registry entry --
2022-01-28 13:29:39.940   Key: HKEY_CURRENT_USER\Environment
2022-01-28 13:29:39.940   Value name: IDF_TOOLS_PATH
2022-01-28 13:29:39.940   Creating or opening the key.
2022-01-28 13:29:39.940   Deleting the value.
2022-01-28 13:29:39.940   Creating or setting the value.
2022-01-28 13:29:39.940   Successfully created or set the value.
2022-01-28 13:29:39.941   Saving uninstall information.
2022-01-28 13:29:39.941   Creating new uninstall key: HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Uninstall\{9E068D99-5C4B-4E5F-96A3-B17CF291E6BD}_is1
2022-01-28 13:29:39.941   Writing uninstall key values.
2022-01-28 13:29:39.945   Detected previous administrative 64-bit install? No
2022-01-28 13:29:39.945   Detected previous administrative 32-bit install? No
2022-01-28 13:29:39.988   Installation process succeeded.
2022-01-28 13:29:39.991   Extracting temporary file: C:\Users\ADMINI~1\AppData\Local\Temp\is-EN362.tmp\7za.exe
2022-01-28 13:29:40.862   Done, exit code=0
2022-01-28 13:29:40.862   --------
2022-01-28 13:29:40.862   Running command: C:\Users\ADMINI~1\AppData\Local\Temp\is-EN362.tmp\idf-env.exe driver install  --ftdi --silabs --espressif
                          Using cached directory: C:\eps\.espressif/tools/idf-driver/silabs-2021-05-0
                          Using cached directory: C:\eps\.espressif/tools/idf-driver/ftdi-2021-05-0
                          Using cached directory: C:\eps\.espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-1
                          Installing driver with INF C:\eps\.espressif\tools\idf-driver\silabs-2021-05-03\silabser.inf -> C:\windows\INF\oem71.inf ... Already installe
                          Installing driver with INF C:\eps\.espressif\tools\idf-driver\ftdi-2021-05-03\ftdiport.inf -> C:\windows\INF\oem57.inf ... Already installe
                          Installing driver with INF C:\eps\.espressif\tools\idf-driver\idf-driver-esp32-usb-jtag-2021-07-15\usb_jtag_debug_unit.inf -> C:\windows\INF\oem72.inf ... Already installe
                          
2022-01-28 13:29:40.862   --------
2022-01-28 13:29:40.875   Setting PATH for this process: C:\eps\.espressif\tools\idf-python\3.8.7\;C:\eps\.espressif\tools\idf-git\2.30.1\cmd\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\windows\System32\OpenSSH\;C:\Program Files\TortoiseSVN\bin;C:\eps\Git\cmd;C:\Users\Administrator\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Bandizip\;C:\Users\Administrator\AppData\Local\GitHubDesktop\bin
2022-01-28 13:29:40.875   PYTHONNOUSERSITE=True
2022-01-28 13:29:40.875   PYTHONPATH=
2022-01-28 13:29:40.875   PYTHONHOME=
2022-01-28 13:29:40.875   IDFTempPath - location of bundle: C:\eps\.espressif\releases\esp-idf-bundle
2022-01-28 13:29:40.876   Updating submodules: C:\eps\.espressif\tools\idf-git\2.30.1\cmd\git.exe -C C:\esp\esp-idf checkout v4.3.1
2022-01-28 13:29:54.858   Done, exit code=0
2022-01-28 13:29:54.858   --------
2022-01-28 13:29:54.858   Running command: C:\eps\.espressif\tools\idf-git\2.30.1\cmd\git.exe -C C:\esp\esp-idf checkout v4.3.1
                          HEAD is now at 2e7491405 versions: Update version to 4.3.
                          
2022-01-28 13:29:54.858   --------
2022-01-28 13:29:54.871   Updating submodules: C:\eps\.espressif\tools\idf-git\2.30.1\cmd\git.exe -C C:\esp\esp-idf submodule update --init --recursive
2022-01-28 13:29:57.081   Done, exit code=1
2022-01-28 13:29:57.081   --------
2022-01-28 13:29:57.081   Running command: C:\eps\.espressif\tools\idf-git\2.30.1\cmd\git.exe -C C:\esp\esp-idf submodule update --init --recursive
                          /mingw64/libexec/git-core/git-sh-setup: line 46: /git-sh-i18n: No such file or director
                          
2022-01-28 13:29:57.081   --------
2022-01-28 13:29:57.081   Message box (OK):
                          Installation has failed with exit code 1
2022-01-28 13:31:05.551   User chose OK.
2022-01-28 13:31:05.570   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2022-01-28 13:31:06.780   User chose Yes.
2022-01-28 13:31:06.872   CurStepChanged raised an exception.
2022-01-28 13:31:07.305   Need to restart Windows? No
2022-01-28 13:31:07.315   -- Run entry --
2022-01-28 13:31:07.315   Run as: Original user
2022-01-28 13:31:07.315   Type: ShellExec
2022-01-28 13:31:07.315   Filename: 
2022-01-28 13:31:07.362   -- Run entry --
2022-01-28 13:31:07.362   Run as: Original user
2022-01-28 13:31:07.362   Type: ShellExec
2022-01-28 13:31:07.362   Filename: 
2022-01-28 13:31:07.474   Deinitializing Setup.
2022-01-28 13:31:07.487   Failed to remove temporary directory: C:\Users\ADMINI~1\AppData\Local\Temp\is-EN362.tmp
2022-01-28 13:31:07.536   Log closed.

georgik, 2022-02-01:
@dasom96 Thank you for reporting the problem. Please make sure that you do not have an obsolete version of git in the PATH. That might cause the problem. Please type `git --version` to determine the version on the PATH.

Please try new release of Online Installer 2.13 which contains update version of git or use one of Offline installers which were published this week:
https://github.com/espressif/idf-installer#table-with-downloads

dasom96, 2022-02-03:
Hello, thank you for your reply.
I reinstalled it offline after removing the installed Git, but it can not be installed.

2022-02-03 10:25:56.619   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\_locale.h
2022-02-03 10:25:56.619   Time stamp of our file: 2022-01-27 23:45:06.000
2022-02-03 10:25:56.619   Installing the file.
2022-02-03 10:25:56.622   Successfully installed the file.
2022-02-03 10:25:56.622   -- File entry --
2022-02-03 10:25:56.624   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\_pthreadtypes.h
2022-02-03 10:25:56.624   Time stamp of our file: 2022-01-27 23:45:06.000
2022-02-03 10:25:56.624   Installing the file.
2022-02-03 10:25:56.628   Successfully installed the file.
2022-02-03 10:25:56.628   -- File entry --
2022-02-03 10:25:56.630   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\_sigset.h
2022-02-03 10:25:56.630   Time stamp of our file: 2022-01-27 23:45:06.000
2022-02-03 10:25:56.630   Installing the file.
2022-02-03 10:25:56.633   Successfully installed the file.
2022-02-03 10:25:56.633   -- File entry --
2022-02-03 10:25:56.635   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\_stdint.h
2022-02-03 10:25:56.635   Time stamp of our file: 2022-01-27 23:45:06.000
2022-02-03 10:25:56.635   Installing the file.
2022-02-03 10:25:56.638   Successfully installed the file.
2022-02-03 10:25:56.639   -- File entry --
2022-02-03 10:25:56.640   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\_timespec.h
2022-02-03 10:25:56.641   Time stamp of our file: 2022-01-27 23:45:06.000
2022-02-03 10:25:56.641   Installing the file.
2022-02-03 10:25:56.644   Successfully installed the file.
2022-02-03 10:25:56.644   -- File entry --
2022-02-03 10:25:56.646   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\_timeval.h
2022-02-03 10:25:56.646   Time stamp of our file: 2022-01-27 23:45:06.000
2022-02-03 10:25:56.646   Installing the file.
2022-02-03 10:25:56.649   Successfully installed the file.
2022-02-03 10:25:56.649   -- File entry --
2022-02-03 10:25:56.651   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\_types.h
2022-02-03 10:25:56.651   Time stamp of our file: 2022-01-27 23:45:06.000
2022-02-03 10:25:56.651   Installing the file.
2022-02-03 10:25:56.654   Successfully installed the file.
2022-02-03 10:25:56.654   -- File entry --
2022-02-03 10:25:56.656   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\xtensa\config\core-isa.h
2022-02-03 10:25:56.657   Time stamp of our file: 2022-01-27 23:45:06.000
2022-02-03 10:25:56.657   Installing the file.
2022-02-03 10:25:56.657   Creating directory: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\xtensa
2022-02-03 10:25:56.657   Creating directory: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\xtensa\config
2022-02-03 10:25:56.662   Successfully installed the file.
2022-02-03 10:25:56.665   -- Registry entry --
2022-02-03 10:25:56.665   Key: HKEY_CURRENT_USER\Environment
2022-02-03 10:25:56.665   Value name: IDF_TOOLS_PATH
2022-02-03 10:25:56.665   Creating or opening the key.
2022-02-03 10:25:56.665   Deleting the value.
2022-02-03 10:25:56.665   Creating or setting the value.
2022-02-03 10:25:56.665   Successfully created or set the value.
2022-02-03 10:25:56.666   Saving uninstall information.
2022-02-03 10:25:56.666   Deleting uninstall key left over from previous non administrative install.
2022-02-03 10:25:56.667   Creating new uninstall key: HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Uninstall\{9E068D99-5C4B-4E5F-96A3-B17CF291E6BD}_is1
2022-02-03 10:25:56.668   Writing uninstall key values.
2022-02-03 10:25:56.668   Detected previous administrative 64-bit install? No
2022-02-03 10:25:56.668   Detected previous administrative 32-bit install? No
2022-02-03 10:25:56.744   Installation process succeeded.
2022-02-03 10:25:56.747   Extracting temporary file: C:\Users\ADMINI~1\AppData\Local\Temp\is-UF591.tmp\7za.exe
2022-02-03 10:25:57.697   Done, exit code=0
2022-02-03 10:25:57.697   --------
2022-02-03 10:25:57.697   Running command: C:\Users\ADMINI~1\AppData\Local\Temp\is-UF591.tmp\idf-env.exe driver install  --ftdi --silabs --espressif
                          Using cached directory: C:\Espressif/tools/idf-driver/silabs-2021-05-0
                          Using cached directory: C:\Espressif/tools/idf-driver/ftdi-2021-05-0
                          Using cached directory: C:\Espressif/tools/idf-driver/idf-driver-esp32-usb-jtag-2021-07-1
                          Installing driver with INF C:\Espressif\tools\idf-driver\silabs-2021-05-03\silabser.inf -> C:\windows\INF\oem71.inf ... Already installe
                          Installing driver with INF C:\Espressif\tools\idf-driver\ftdi-2021-05-03\ftdiport.inf -> C:\windows\INF\oem57.inf ... Already installe
[... cut ...]
2022-02-03 10:45:42.940   -- Run entry --
2022-02-03 10:45:42.940   Run as: Original user
2022-02-03 10:45:42.940   Type: ShellExec
2022-02-03 10:45:42.940   Filename: 
2022-02-03 10:45:42.977   -- Run entry --
2022-02-03 10:45:42.977   Run as: Original user
2022-02-03 10:45:42.977   Type: ShellExec
2022-02-03 10:45:42.977   Filename: 
2022-02-03 10:45:43.080   Deinitializing Setup.
2022-02-03 10:45:43.092   Failed to remove temporary directory: C:\Users\ADMINI~1\AppData\Local\Temp\is-UF591.tmp
2022-02-03 10:45:43.112   Log closed.

dasom96, 2022-02-03:
@georgik 

Only Offline can be installed due to the security policy of the company PC.
I checked if it was because of the company's security network, but all my co-workers succeeded in installing it offline.
I'm the only one who can't install as above.

[Single ESP-IDF Branch Offline Installer]
I tried both ESP-IDF v4.4 and v4.3.2, but all of them failed to install.

georgik, 2022-02-03:
Thank you @dasom96 for the log.

The Offline installer should contain all the content and subsequent `idf.py install` should not download anything from the Internet.

The strange thing is this warning:
`WARNING: tool xtensa-esp32s2-elf version esp-2021r2-8.4.0 is installed, but has reported version unknown`

The toolchain should be already deployed at `C:\Espressif\tools`.
Please, can you check whether directory for xtensa-esp32s2-elf exists and try to execute following command in the directory to acquire the version:
```
xtensa-esp32s2-elf-gcc --version
```

georgik, 2022-02-03:
@dasom96 One alternative solution might be to use Online installer and to check following options in Component Selection Screen (bottom of the list):
- Optimization
  - Use Gitee mirror
  - Use Espressif download server instead of downloading tool packages from GitHub.

georgik, 2022-02-03:
@dasom96 We made additional tests of the installer and we were not able to simulate the problem.
Please could you post the information from System Check page of the installer?
It might be caused by an antivirus. Do you have Comodo? The information is visible at the System Check page (second screen of the installer).

dasom96, 2022-02-08:
@georgik 
Below is the system check page of the online installation program (Full log).

2022-02-08 09:17:07.512   Log opened. (Time zone: UTC+09:00)
2022-02-08 09:17:07.512   Setup version: Inno Setup version 6.2.0
2022-02-08 09:17:07.512   Original Setup EXE: C:\Users\Administrator\Downloads\esp-idf-tools-setup-online-2.12.exe
2022-02-08 09:17:07.512   Setup command line: /SL5="$5B0890,2558306,831488,C:\Users\Administrator\Downloads\esp-idf-tools-setup-online-2.12.exe" 
2022-02-08 09:17:07.512   Windows version: 10.0.18363  (NT platform: Yes)
2022-02-08 09:17:07.512   64-bit Windows: Yes
2022-02-08 09:17:07.512   Processor architecture: x64
2022-02-08 09:17:07.512   User privileges: Administrative
2022-02-08 09:17:07.516   Administrative install mode: No
2022-02-08 09:17:07.516   Install mode root key: HKEY_CURRENT_USER
2022-02-08 09:17:07.516   64-bit install mode: Yes
2022-02-08 09:17:08.502   Created temporary directory: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp
2022-02-08 09:17:08.510   -- DLL function import --
2022-02-08 09:17:08.510   Function name: idpAddFile
2022-02-08 09:17:08.510   DLL name: files:idp.dll
2022-02-08 09:17:08.510   Extracting temporary file: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idp.dll
2022-02-08 09:17:08.518   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idp.dll
2022-02-08 09:17:08.518   Importing the DLL function.
2022-02-08 09:17:08.522   Successfully imported the DLL function. Delay loaded? No
2022-02-08 09:17:08.522   -- DLL function import --
2022-02-08 09:17:08.522   Function name: idpAddFileComp
2022-02-08 09:17:08.522   DLL name: files:idp.dll
2022-02-08 09:17:08.522   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idp.dll
2022-02-08 09:17:08.522   Importing the DLL function.
2022-02-08 09:17:08.522   Successfully imported the DLL function. Delay loaded? No
2022-02-08 09:17:08.522   -- DLL function import --
2022-02-08 09:17:08.522   Function name: idpAddMirror
2022-02-08 09:17:08.522   DLL name: files:idp.dll
2022-02-08 09:17:08.523   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idp.dll
2022-02-08 09:17:08.523   Importing the DLL function.
2022-02-08 09:17:08.523   Successfully imported the DLL function. Delay loaded? No
2022-02-08 09:17:08.523   -- DLL function import --
2022-02-08 09:17:08.523   Function name: idpAddFtpDir
2022-02-08 09:17:08.523   DLL name: files:idp.dll
2022-02-08 09:17:08.523   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idp.dll
2022-02-08 09:17:08.523   Importing the DLL function.
2022-02-08 09:17:08.523   Successfully imported the DLL function. Delay loaded? No
2022-02-08 09:17:08.523   -- DLL function import --
2022-02-08 09:17:08.523   Function name: idpAddFtpDirComp
2022-02-08 09:17:08.523   DLL name: files:idp.dll
2022-02-08 09:17:08.523   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idp.dll
2022-02-08 09:17:08.523   Importing the DLL function.
2022-02-08 09:17:08.523   Successfully imported the DLL function. Delay loaded? No
2022-02-08 09:17:08.523   -- DLL function import --
2022-02-08 09:17:08.523   Function name: idpClearFiles
2022-02-08 09:17:08.523   DLL name: files:idp.dll
2022-02-08 09:17:08.523   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idp.dll
2022-02-08 09:17:08.523   Importing the DLL function.
2022-02-08 09:17:08.523   Successfully imported the DLL function. Delay loaded? No
2022-02-08 09:17:08.523   -- DLL function import --
2022-02-08 09:17:08.523   Function name: idpFilesCount
2022-02-08 09:17:08.523   DLL name: files:idp.dll
2022-02-08 09:17:08.523   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idp.dll
2022-02-08 09:17:08.523   Importing the DLL function.
2022-02-08 09:17:08.523   Successfully imported the DLL function. Delay loaded? No
2022-02-08 09:17:08.523   -- DLL function import --
2022-02-08 09:17:08.523   Function name: idpFtpDirsCount
2022-02-08 09:17:08.523   DLL name: files:idp.dll
2022-02-08 09:17:08.523   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idp.dll
2022-02-08 09:17:08.523   Importing the DLL function.
2022-02-08 09:17:08.523   Successfully imported the DLL function. Delay loaded? No
2022-02-08 09:17:08.523   -- DLL function import --
2022-02-08 09:17:08.523   Function name: idpFileDownloaded
2022-02-08 09:17:08.523   DLL name: files:idp.dll
2022-02-08 09:17:08.523   Dest DLL name: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idp.dll
2022-02-08 09:17:08.523   Importing the DLL function.
2022-02-08 09:17:08.523   Successfully imported the DLL function. Delay loaded? No
[... 200 lines cut ...]
2022-02-08 09:17:11.000   * Starting system check ...
2022-02-08 09:17:11.009   * Windows version
2022-02-08 09:17:11.010   : 10.00.18363
2022-02-08 09:17:11.012    [OK]
2022-02-08 09:17:11.013   * Checking "Long Paths Enabled" in Windows registry 
2022-02-08 09:17:11.015    [OK]
2022-02-08 09:17:11.016   * Active code page: 
2022-02-08 09:17:11.017   Executing: chcp.com
2022-02-08 09:17:11.352   
2022-02-08 09:17:11.355   * Checking certificates 
2022-02-08 09:17:11.357    $ C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idf-env.exe certificate verify --url https://dl.espressif.com/dl/?system_check=win10.00.18363
2022-02-08 09:17:12.776    $ C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idf-env.exe certificate verify --url https://github.com/espressif
[... cut ...]
2022-02-08 09:17:37.403   Executing: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idf-env.exe antivirus get --property displayName
2022-02-08 09:17:38.498   Result: AhnLab V3 Internet Security 9.0
2022-02-08 09:17:38.498   Executing: C:\Users\ADMINI~1\AppData\Local\Temp\is-5GO3B.tmp\idf-env.exe antivirus get --property displayName
2022-02-08 09:17:39.498   Result: AhnLab V3 Internet Security 9.0
2022-02-08 09:17:39.498   * Detected antivirus: AhnLab V3 Internet Security 9.0
2022-02-08 09:17:39.504   * Check complete.

dasom96, 2022-02-08:
Below is the system check page of the offline installation program.

* Starting system check ...
* Windows version: 10.00.18363 [OK]
* Checking "Long Paths Enabled" in Windows registry  [OK]
* Active code page: 
* Checking certificates .. [WARN]Unable to load data from server dl.espressif.com.
* Detected antivirus: AhnLab V3 Internet Security 9.0
* Check complete.

dasom96, 2022-02-08:
I found xtensa-esp32s2-elf directory, but version acquisition failed. The command is not executed.

Is there a way to download each necessary program, not an integrated installation program?

georgik, 2022-02-08:
Hi @dasom96 . Thank you for detailed report. It seems that s3.amazonws.com is blocked.

Topic 1: Interestingly enough Offline installer does not work for you. I'm still wondering why it tries to download anything from Internet. Please could you try to run the Offline installer when the computer is completely disconnected?

Topic 2: Yes, it's possible to download all the tools on other computer. Simply run the installer on the computer with network connectivity and then copy .espressif\tools directory or .espressif\dist (which is the same, just it contains compressed version of tools which is then expanded to tools)

Topic 3: If previous approach fails. You might consider downloading files manually. URLs are located here: https://github.com/espressif/esp-idf/blob/master/tools/tools.json

Topic 4: It might be worthy to add "Extract only" option to the Offline Installer to avoid download failures. This might take a day to implement.

Please, let me know if any of these options works for you.

georgik, 2022-02-08:
The problem might be caused by the antivirus. My guess is that the antivirus blocks execution of this binary `xtensa-esp32s2-elf-gcc` which results in idf_tools.py interpreting the error as unknown version and therefore it starts installing the new version from online storage which fails.

georgik, 2022-02-08:
@dasom96 We were not able to install AhnLab antivirus. Windows reported the software as causing severe security and performance issues both on Windows 10 and Windows 11.

Please, contact your IT support and ask them to exclude directory C:\Espressif from blocking/scanning. The tools like xtensa-esp32s2-elf-gcc are required to run to compile the software. Without them it's not possible to compile the binary for ESP.

One alternative approach for Windows users is to install the toolchain to WSL2 with Ubuntu 20 and use Visual Studio Code with remote WSL support to connect to VM in the WSL2.

dasom96, 2022-02-10:
I reinstalled it after excluding V3 inspection, but it failed.

2022-02-10 16:31:15.373   Successfully installed the file.
2022-02-10 16:31:15.374   -- File entry --
2022-02-10 16:31:15.375   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\features.h
2022-02-10 16:31:15.376   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.376   Installing the file.
2022-02-10 16:31:15.377   Successfully installed the file.
2022-02-10 16:31:15.378   -- File entry --
2022-02-10 16:31:15.379   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\fenv.h
2022-02-10 16:31:15.379   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.379   Installing the file.
2022-02-10 16:31:15.381   Successfully installed the file.
2022-02-10 16:31:15.381   -- File entry --
2022-02-10 16:31:15.383   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\file.h
2022-02-10 16:31:15.383   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.383   Installing the file.
2022-02-10 16:31:15.384   Successfully installed the file.
2022-02-10 16:31:15.385   -- File entry --
2022-02-10 16:31:15.386   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\iconvnls.h
2022-02-10 16:31:15.387   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.387   Installing the file.
2022-02-10 16:31:15.388   Successfully installed the file.
2022-02-10 16:31:15.389   -- File entry --
2022-02-10 16:31:15.391   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\lock.h
2022-02-10 16:31:15.391   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.391   Installing the file.
2022-02-10 16:31:15.392   Successfully installed the file.
2022-02-10 16:31:15.393   -- File entry --
2022-02-10 16:31:15.394   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\param.h
2022-02-10 16:31:15.395   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.395   Installing the file.
2022-02-10 16:31:15.396   Successfully installed the file.
2022-02-10 16:31:15.396   -- File entry --
2022-02-10 16:31:15.398   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\queue.h
2022-02-10 16:31:15.398   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.398   Installing the file.
2022-02-10 16:31:15.400   Successfully installed the file.
2022-02-10 16:31:15.400   -- File entry --
2022-02-10 16:31:15.402   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\reent.h
2022-02-10 16:31:15.402   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.402   Installing the file.
2022-02-10 16:31:15.404   Successfully installed the file.
2022-02-10 16:31:15.405   -- File entry --
2022-02-10 16:31:15.406   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\resource.h
2022-02-10 16:31:15.407   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.407   Installing the file.
2022-02-10 16:31:15.408   Successfully installed the file.
2022-02-10 16:31:15.409   -- File entry --
2022-02-10 16:31:15.410   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\sched.h
2022-02-10 16:31:15.410   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.411   Installing the file.
2022-02-10 16:31:15.412   Successfully installed the file.
2022-02-10 16:31:15.412   -- File entry --
2022-02-10 16:31:15.414   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\select.h
2022-02-10 16:31:15.414   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.414   Installing the file.
2022-02-10 16:31:15.415   Successfully installed the file.
2022-02-10 16:31:15.415   -- File entry --
2022-02-10 16:31:15.418   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\signal.h
2022-02-10 16:31:15.418   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.418   Installing the file.
2022-02-10 16:31:15.420   Successfully installed the file.
2022-02-10 16:31:15.420   -- File entry --
2022-02-10 16:31:15.422   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\stat.h
2022-02-10 16:31:15.422   Time stamp of our file: 2022-01-28 18:14:02.000
2022-02-10 16:31:15.422   Installing the file.
2022-02-10 16:31:15.423   Successfully installed the file.
2022-02-10 16:31:15.424   -- File entry --
2022-02-10 16:31:15.426   Dest filename: \\?\C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\stdio.h
[... 175 lines cut ...]
[... cut ...]
2022-02-10 16:35:55.469   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2022-02-10 16:35:57.119   User chose Yes.
2022-02-10 16:35:57.218   CurStepChanged raised an exception.
2022-02-10 16:35:57.632   Need to restart Windows? No
2022-02-10 16:35:57.640   -- Run entry --
2022-02-10 16:35:57.640   Run as: Original user
2022-02-10 16:35:57.640   Type: ShellExec
2022-02-10 16:35:57.640   Filename:

dasom96, 2022-02-17:
@georgik Hello, 

The company's IT support team says it is no longer possible to take action against AhnLab.

So i downloaded esp-idf to my personal computer and transferred all the files to my company computer.
It took a long time to transferred the file, but it eventually became executable.

Thank you for your response.

georgik, 2022-02-17:
Thank you @dasom96 very much for update. I'm curious why it's possible to copy the file from other computer, but the download from internet fails.
Please, can you check whether file is not marked as "Blocked": right click the file, Properties, at the bottom you might see "Security" field and checkbox "Unblock".
[image]
Theoretically the file downloaded as ZIP might be marked as blocked and the antivirus blocks the execution. Having this information might help us to detect the problem on installer level