espressif/idf-installer #119: ZIP releases of ESP-IDF should be shallow

georgik, 2022-03-08:
ZIP releases of ESP-IDF is not shallow, which adds several hundred megabytes to download and storage space.

Update build of ZIP so, that it corresponds to Offline installers.

Update steps:  https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#updating-esp-idf

igrr, 2022-03-08:
For future reference, the line which is responsible for cloning the repository during zip file creation is here: https://github.com/espressif/github-actions/blob/a9f8c9761ada5717ff05e2ebd3791b0328c9d21d/release_zips/release_zips.py#L39

georgik, 2023-02-01:
Update: use `--single-branch --shallow-submodules` instead of `--depth 1` to keep `git describe` working.