espressif/idf-installer #124: Espressif IDE Java not finding

franz-ms-muc, 2022-03-09:
the Amazon JDK is in 
espressif\tools\amazon-corretto-11-x64-windows-jdk
the IDE in 
espressif\tools\espressif-ide\2.4.0\

but it show this Error. 

[image]

either put the jre in the subfolder of the eclipse, 
or set the according -vm Argument, 
or set a PATH to the corretto. (problem if other Eclipse rely on this, 
as Coretto is causing Problems there)

kolipakakondal, 2022-03-09:
Hi @franz-ms-muc Thanks for reporting. Let us look into this.

georgik, 2022-03-10:
@franz-ms-muc Thank you for reporting the problem.
The path is not set correctly, because it's fixed to previous minor release of JDK.
Workaround: add path manually.

The new build with the correction should be available shortly.

georgik, 2022-03-10:
@franz-ms-muc Please re-download new build of the installer from https://github.com/espressif/idf-installer/releases/tag/espressif-ide-2.4.2
The Path environment variable should be updated correctly.

PR #125 (2022-03-10): tools: fix installation of jdk11.0.14_10
Closes #124
closed 2022-03-10 by georgik