espressif/idf-installer #131: Installation has failed with exit code -02

TheButlah, 2022-04-08:
I selected only the drivers in the esp-idf installer. This is on v4.4 offline installer, windows 10.

```
2022-04-08 04:32:08.366   Log opened. (Time zone: UTC-04:00)
2022-04-08 04:32:08.367   Setup version: Inno Setup version 6.2.0
2022-04-08 04:32:08.367   Original Setup EXE: D:\Ryan\Downloads\esp-idf-tools-setup-offline-4.4.exe
2022-04-08 04:32:08.367   Setup command line: /SL5="$E060E,627157437,843264,D:\Ryan\Downloads\esp-idf-tools-setup-offline-4.4.exe" 
2022-04-08 04:32:08.367   Windows version: 10.0.19044  (NT platform: Yes)
2022-04-08 04:32:08.367   64-bit Windows: Yes
2022-04-08 04:32:08.367   Processor architecture: x64
2022-04-08 04:32:08.367   User privileges: None
2022-04-08 04:32:08.464   Administrative install mode: No
2022-04-08 04:32:08.464   Install mode root key: HKEY_CURRENT_USER
2022-04-08 04:32:08.464   64-bit install mode: Yes
2022-04-08 04:32:10.699   Created temporary directory: C:\Users\Ryan\AppData\Local\Temp\is-I74F5.tmp
2022-04-08 04:32:10.703   -- DLL function import --
2022-04-08 04:32:10.703   Function name: idpAddFile
2022-04-08 04:32:10.704   DLL name: files:idp.dll
2022-04-08 04:32:10.704   Extracting temporary file: C:\Users\Ryan\AppData\Local\Temp\is-I74F5.tmp\idp.dll
2022-04-08 04:32:10.709   Dest DLL name: C:\Users\Ryan\AppData\Local\Temp\is-I74F5.tmp\idp.dll
2022-04-08 04:32:10.709   Importing the DLL function.
2022-04-08 04:32:10.738   Successfully imported the DLL function. Delay loaded? No
2022-04-08 04:32:10.738   -- DLL function import --
2022-04-08 04:32:10.738   Function name: idpAddFileComp
2022-04-08 04:32:10.738   DLL name: files:idp.dll
2022-04-08 04:32:10.738   Dest DLL name: C:\Users\Ryan\AppData\Local\Temp\is-I74F5.tmp\idp.dll
2022-04-08 04:32:10.738   Importing the DLL function.
2022-04-08 04:32:10.738   Successfully imported the DLL function. Delay loaded? No
2022-04-08 04:32:10.738   -- DLL function import --
2022-04-08 04:32:10.738   Function name: idpAddMirror
2022-04-08 04:32:10.738   DLL name: files:idp.dll
2022-04-08 04:32:10.739   Dest DLL name: C:\Users\Ryan\AppData\Local\Temp\is-I74F5.tmp\idp.dll
2022-04-08 04:32:10.739   Importing the DLL function.
2022-04-08 04:32:10.739   Successfully imported the DLL function. Delay loaded? No
2022-04-08 04:32:10.739   -- DLL function import --
2022-04-08 04:32:10.739   Function name: idpAddFtpDir
2022-04-08 04:32:10.739   DLL name: files:idp.dll
2022-04-08 04:32:10.739   Dest DLL name: C:\Users\Ryan\AppData\Local\Temp\is-I74F5.tmp\idp.dll
2022-04-08 04:32:10.739   Importing the DLL function.
2022-04-08 04:32:10.739   Successfully imported the DLL function. Delay loaded? No
2022-04-08 04:32:10.739   -- DLL function import --
2022-04-08 04:32:10.739   Function name: idpAddFtpDirComp
2022-04-08 04:32:10.739   DLL name: files:idp.dll
2022-04-08 04:32:10.739   Dest DLL name: C:\Users\Ryan\AppData\Local\Temp\is-I74F5.tmp\idp.dll
2022-04-08 04:32:10.739   Importing the DLL function.
2022-04-08 04:32:10.739   Successfully imported the DLL function. Delay loaded? No
2022-04-08 04:32:10.739   -- DLL function import --
2022-04-08 04:32:10.739   Function name: idpClearFiles
2022-04-08 04:32:10.739   DLL name: files:idp.dll
2022-04-08 04:32:10.739   Dest DLL name: C:\Users\Ryan\AppData\Local\Temp\is-I74F5.tmp\idp.dll
2022-04-08 04:32:10.739   Importing the DLL function.
2022-04-08 04:32:10.739   Successfully imported the DLL function. Delay loaded? No
2022-04-08 04:32:10.739   -- DLL function import --
2022-04-08 04:32:10.739   Function name: idpFilesCount
2022-04-08 04:32:10.739   DLL name: files:idp.dll
2022-04-08 04:32:10.739   Dest DLL name: C:\Users\Ryan\AppData\Local\Temp\is-I74F5.tmp\idp.dll
2022-04-08 04:32:10.739   Importing the DLL function.
2022-04-08 04:32:10.739   Successfully imported the DLL function. Delay loaded? No
2022-04-08 04:32:10.739   -- DLL function import --
2022-04-08 04:32:10.739   Function name: idpFtpDirsCount
2022-04-08 04:32:10.739   DLL name: files:idp.dll
2022-04-08 04:32:10.739   Dest DLL name: C:\Users\Ryan\AppData\Local\Temp\is-I74F5.tmp\idp.dll
2022-04-08 04:32:10.739   Importing the DLL function.
2022-04-08 04:32:10.739   Successfully imported the DLL function. Delay loaded? No
2022-04-08 04:32:10.739   -- DLL function import --
2022-04-08 04:32:10.739   Function name: idpFileDownloaded
2022-04-08 04:32:10.739   DLL name: files:idp.dll
2022-04-08 04:32:10.739   Dest DLL name: C:\Users\Ryan\AppData\Local\Temp\is-I74F5.tmp\idp.dll
2022-04-08 04:32:10.739   Importing the DLL function.
2022-04-08 04:32:10.739   Successfully imported the DLL function. Delay loaded? No
[... 359 lines cut ...]
2022-04-08 04:33:10.391   idf_tools.py does not exist in IDF directory, using a fallback version
2022-04-08 04:33:10.391   Offline installation selected. Setting environment variable PIP_NO_INDEX=1
2022-04-08 04:33:10.391   idf_tools.py command: D:\Espressif\tools\idf-python\3.8.7\python.exe "D:\Espressif\idf_tools_fallback.py" --idf-path D:\Espressif\frameworks\esp-idf-v4.4\ --tools "D:\Espressif\tools_fallback.json" 
2022-04-08 04:33:10.391   Selection of targets testing command: D:\Espressif\tools\idf-python\3.8.7\python.exe "D:\Espressif\idf_tools_fallback.py" install --targets=""
2022-04-08 04:33:10.399   Selection of targets: Not supported in this version idf_tools.py
2022-04-08 04:33:10.399   Installing tools:D:\Espressif\tools\idf-python\3.8.7\python.exe "D:\Espressif\idf_tools_fallback.py" --idf-path D:\Espressif\frameworks\esp-idf-v4.4\ --tools "D:\Espressif\tools_fallback.json"  install
2022-04-08 04:33:10.429   ProcStart failed
2022-04-08 04:33:10.429   Done, exit code=-2
2022-04-08 04:33:10.429   --------
2022-04-08 04:33:10.429   Running command: D:\Espressif\tools\idf-python\3.8.7\python.exe "D:\Espressif\idf_tools_fallback.py" --idf-path D:\Espressif\frameworks\esp-idf-v4.4\ --tools "D:\Espressif\tools_fallback.json"  install
                          
2022-04-08 04:33:10.429   --------
2022-04-08 04:33:10.429   Message box (OK):
                          Installation has failed with exit code -2
2022-04-08 04:33:16.044   User chose OK.
2022-04-08 04:33:16.052   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2022-04-08 04:33:18.749   User chose Yes.
2022-04-08 04:33:18.785   CurStepChanged raised an exception.
2022-04-08 04:33:18.819   Need to restart Windows? No
2022-04-08 04:33:18.824   Deinitializing Setup.
2022-04-08 04:33:18.827   Failed to remove temporary directory: C:\Users\Ryan\AppData\Local\Temp\is-I74F5.tmp
2022-04-08 04:33:18.838   Log closed.
```

georgik, 2022-04-11:
Thank you @TheButlah for reporting the problem. The drivers was installed, just the installer exists with error.

The installation of only drivers is not suported yet: https://github.com/espressif/idf-installer/issues/113

georgik, 2022-04-11:
@TheButlah Please test new version of installer: https://github.com/espressif/idf-installer/releases/tag/online-2.15-beta-01
This version should allow installation when drivers are the only selected component.

TheButlah, 2022-04-11:
I think that worked. Hard to actually know because I still have https://github.com/espressif/openocd-esp32/issues/222

PR #132 (2022-04-19): tools: allow installation of drivers without rest of the stack
Closes #131
Closes #113

Installation of drivers component without any other component.
closed 2022-04-26 by georgik