espressif/idf-installer #142: ESP-IDF v5 and VisualGDB - No module named idf_component_manager

gojimmypi, 2022-07-09:
I was hoping to use ESP-IDF version 5.x in VisualGDB, mainly for the robust JTAG single step & breakpoint capabilities in Visual Studio. This error is occurring:

```
CMake Error at C:/SysGCC/esp32/esp-idf/v5.0/tools/cmake/build.cmake:525 (message):
  C:\SysGCC\esp32\esp-idf\v5.0\python-env\Scripts\python.exe: Error while
  finding module specification for 'idf_component_manager.prepare_components'
  (ModuleNotFoundError: No module named 'idf_component_manager')
```
## Background

I do see although there's a [release/v5.0 branch](https://github.com/espressif/esp-idf/tree/release/v5.0) it is _not_ listed in the [release list](https://github.com/espressif/esp-idf/releases) nor even shown in the [release support schedule](https://github.com/espressif/esp-idf#esp-idf-release-support-schedule), but v5 _is_ listed in [ESP-IDF Release and SoC Compatibility](https://github.com/espressif/esp-idf#esp-idf-release-and-soc-compatibility) chart. 

I did reach out on the [VisualGDB forum](https://sysprogs.com/w/forums/topic/esp-idf-v5/), but of course they are not yet supporting a version that is not yet officially released. 

This single line of text from [ESP-IDF and Toolchains](https://visualgdb.com/documentation/espidf/#idf) has me remaining hopeful this is possible:

> Any project that can be built via ESP-IDF tools, can also be built with VisualGDB.

## Install

Still, I am optimistic, as I have it _really close_ to working. I'm doing this on Windows 10:

```
cd C:\SysGCC\esp32\esp-idf
git clone -b release/v5.0 --recursive https://github.com/espressif/esp-idf.git v5.0
cd v5.0
install.bat all
export.bat
```
## Build with `idf.py` success

I have a [project created with the VisualGDB Wizard](https://github.com/gojimmypi/esp-idf-v5-blinky) that successfully builds and flashes for the ESP32-C3 V2 using the `idf.py build`:

```
C:\SysGCC\esp32\esp-idf\v5.0\export.bat
```

with this output:

```
Checking Python compatibility
Setting IDF_PATH: C:\SysGCC\esp32\esp-idf\v5.0

Adding ESP-IDF tools to PATH...
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.23.1.
Not using an unsupported version of tool ninja found in PATH: 1.8.2.
    C:\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220529\xtensa-esp-elf-gdb\bin
    C:\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220529\riscv32-esp-elf-gdb\bin
    C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-RC1-11.2.0\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-2022r1-RC1-11.2.0\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-RC1-11.2.0\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-2022r1-RC1-11.2.0\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Espressif\python_env\idf5.0_py3.8_env\Scripts
    C:\SysGCC\esp32\esp-idf\v5.0\tools

Checking if Python packages are up to date...
Skipping the download of C:\Espressif\espidf.constraints.v5.0.txt because it was downloaded recently. If you believe that this is causing you trouble then remove it manually and re-run your install script.
Constraint file: C:\Espressif\espidf.constraints.v5.0.txt
Requirement files:
 - C:\SysGCC\esp32\esp-idf\v5.0\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing amazon-corretto-11-x64-windows-jdk, dfu-util, esp32s2ulp-elf, esp32ulp-elf, espressif-ide, riscv32-esp-elf, riscv32-esp-elf-gdb, xtensa-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf use command 'python.exe C:\SysGCC\esp32\esp-idf\v5.0\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe C:\SysGCC\esp32\esp-idf\v5.0\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
```

and indeed `idf.py build` is successful:

```
Executing action: all (aliases: build)
Running ninja in directory c:\users\gojimmypi\source\repos\embeddedproject10\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file C:/Users/gojimmypi/source/repos/EmbeddedProject10/sdkconfig

[... snip, etc...]

-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/gojimmypi/source/repos/EmbeddedProject10/build/bootloader
[1/2] Generating binary image from built executable
esptool.py v4.1
Creating esp32c3 image...
Merged 1 ELF section
Successfully created esp32c3 image.
Generated C:/Users/gojimmypi/source/repos/EmbeddedProject10/build/bootloader/bootloader.bin
[2/2] cmd.exe /C "cd /D C:\Users\gojimmypi\source\repos\EmbeddedProject10\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe C:/SysGCC/esp32/esp-idf/v5.0/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/gojimmypi/source/repos/EmbeddedProject10/build/bootloader/bootloader.bin"
Bootloader binary size 0x4e20 bytes. 0x31e0 bytes (39%) free.
[14/14] Completed 'bootloader'

Project build complete. To flash, run this command:
C:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe ..\..\..\..\..\SysGCC\esp32\esp-idf\v5.0\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32c3  write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\blink.bin
or run 'idf.py -p (PORT) flash'
```

## `CMake` Error in VisualGDB
However, when I open the VisualGDB solution file, `cmake` fails with this error:

```
Running CMake: C:\Users\gojimmypi\AppData\Local\VisualGDB\CMake\bin\cmake.exe ../../.. -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_MAKE_PROGRAM=c:/SysGCC/esp32/tools/ninja/1.10.2/ninja.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 -DSDKCONFIG=C:/Users/gojimmypi/source/repos/EmbeddedProject10/sdkconfig
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements are satisfied.
Skipping the download of C:\Users\gojimmypi\.espressif\espidf.constraints.v5.0.txt because it was downloaded recently. If you believe that this is causing you trouble then remove it manually and re-run your install script.
Constraint file: C:\Users\gojimmypi\.espressif\espidf.constraints.v5.0.txt
Requirement files:
 - C:\SysGCC\esp32\esp-idf\v5.0\tools\requirements\requirements.core.txt
Python being checked: C:\Users\gojimmypi\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
-- Configuring incomplete, errors occurred!
See also "C:/Users/gojimmypi/source/repos/EmbeddedProject10/build/VisualGDB/Debug/CMakeFiles/CMakeOutput.log".
CMake Error at C:/SysGCC/esp32/esp-idf/v5.0/tools/cmake/build.cmake:525 (message):
  C:\SysGCC\esp32\esp-idf\v5.0\python-env\Scripts\python.exe: Error while
  finding module specification for 'idf_component_manager.prepare_components'
  (ModuleNotFoundError: No module named 'idf_component_manager')

Call Stack (most recent call first):
  C:/SysGCC/esp32/esp-idf/v5.0/tools/cmake/project.cmake:434 (idf_build_process)
  CMakeLists.txt:10 (project)

System.Exception: CMake exited with code 1
   at a32.i(c1 b, Object a)

========== Project Configuration Summary ==========
    EmbeddedProject10  configured in 00:18
========== Configuration: 1 Succeeded, 0 Failed, 0 Skipped ==========
```

I tried turning off the `idf_component_manager` with the environment variable setting, but I only saw other `cmake` errors.

## VisualGDB setup

Here are my settings in VisualGDB. I manually selected the `release/5.0` directory and browsed to `C:\Users\gojimmypi\source\repos\EmbeddedProject10` to find the `sdkconfig` file that was created by `idf.py`.

[image]

## Environment Versions

[image]

[image]

```
C:\Users\gojimmypi\source\repos\EmbeddedProject10>idf.py --version
ESP-IDF v5.0-dev-3954-g271f4e4c40

C:\Users\gojimmypi\source\repos\EmbeddedProject10>python -V
Python 3.8.7
```

## See also
- VisuaGDB [Advanced ESP-IDF Project Structure](https://visualgdb.com/documentation/espidf/)
- VisuaGDB [Switching Advanced ESP-IDF Projects Between Different IDF Versions](https://visualgdb.com/tutorials/esp32/esp-idf/multiple/)
- Espressif [IDF Component Manager](https://github.com/espressif/idf-component-manager)
-

ESP32DE, 2022-07-09:
@gojimmypi 

idf build for target esp32c3 and sdkconfig is saved for this target esp32-c3
`-- Building ESP-IDF components for target esp32c3`

CMake Error in VisualGDB
`-- Building ESP-IDF components for target esp32`

can you check?

gojimmypi, 2022-07-09:
hi @ESP32DE ! Thanks for your help. The [sdkconfig](https://github.com/gojimmypi/esp-idf-v5-blinky/blob/main/sdkconfig) is in the project directory, in my case `C:/Users/gojimmypi/source/repos/EmbeddedProject10`. This is the file generated by `idf.py`.

I did notice the `esp32` vs `esp32c3`, and added a manual CMake setting in the VisualGDB UI for `IDF_TARGET = esp32c3` (see [update](https://sysprogs.com/w/forums/topic/esp-idf-v5/#post-32812)). I had incorrectly assumed that would have been used as found in the [sdkconfig on line 200](https://github.com/gojimmypi/esp-idf-v5-blinky/blob/2e7e9c8e0f53a68d52d3b84b8fb64b71853e9c5a/sdkconfig#L200).

I'm able to get a modified `CMake` to run successfully at a fresh command prompt (not an ESP-IDF prompt), but I've not yet been able to persuade VisualGDB to generate one exactly like that just yet. It seems at this point to be a matter of environment variables and path.

gojimmypi, 2022-07-14:
thanks everyone for taking a look at this.

>  Isn't it possible to pick the environment deployed by idf-installer in VisualGDB?

@georgik that would be really cool if possible, but I don't think so, and least not in the way I think you are asking. VisualGDB keeps track of its own environments installed in `C:\SysGCC`. In my case, this is `C:\SysGCC\esp8266` and `C:\SysGCC\esp32`. My `C:\SysGCC\esp32\esp-idf` contains:

[image]

And so ya: in VisualGDB I can select between `v4.4.1` that the sysprogs installer created and `v5.0` that I created.  

In all fairness from the [Troubleshooting](https://visualgdb.com/documentation/espidf/) section:

[image]

But in general, I do not think it is possible to switch to the default ESP-IDF directory created by the Espressif ESP-IDF install process from within Visual Studio. I think it would be cool if possible, but they likely install their own so as to keep better control of it. (i.e. easier to support)

Although... in my _particular_ case, I _am_ able to do that, but only as I specified `C:\SysGCC\esp32\esp-idf\v5.0` as the install destination for the ESP-IDF. So my prompt ends up there:

[image]

Regarding my python:

> Do you remember how the python you are using was installed

In this particular case, I remember very well - as I _had_ Python 3.10 dot something installed when I first started with this issue, but I had encountered some path issue that was apparently hardcoded for Python 3.8. Fortunately I had a `python-3.8.1.exe` still in my download folder from the `python.org/downloads`.  That's my currently installed version.

@kumekay It is a really good question though, as I recall some time ago having all sorts of problems with the Visual Studio version of Python installed from the `VS Installer`. I confirmed I _do not_ have an Visual Studio Python components installed:

[image]

Regarding the suggestion from @igrr , a value of 

`C:\Users\gojimmypi\AppData\Local\VisualGDB\Python3`

was in my settings:

[image]

I cleared that value as described in the [espidf python](https://visualgdb.com/documentation/espidf/#python) section, deleted my `%LOCALAPPDATA%\VisualGDB\Python`:

[image]

and then "Get Python 3" from a fresh New Project Wizard (note that I explicitly selected the esp-idf 5 that I installed):

[image]

the installer is really cool. No magic commands, just a button push and tada:

[image]

Unfortunately, I see pretty much the same error at new project wizard time:

```
Running CMake: C:\Users\gojimmypi\AppData\Local\VisualGDB\CMake\bin\cmake.exe ../../.. -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_MAKE_PROGRAM=c:/SysGCC/esp32/tools/ninja/1.10.2/ninja.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=0
-- IDF_TARGET not set, using default target: esp32
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.30.2.windows.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/SysGCC/esp32/tools/xtensa-esp32-elf/esp-2021r2-patch3/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/SysGCC/esp32/tools/xtensa-esp32-elf/esp-2021r2-patch3/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/SysGCC/esp32/tools/xtensa-esp32-elf/esp-2021r2-patch3/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
[... 55 lines cut ...]
Requirement already satisfied: charset-normalizer~=2.0.0 in c:\users\gojimmypi\appdata\local\visualgdb\python3\lib\site-packages (from requests<3->idf-component-manager) (2.0.7)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in c:\users\gojimmypi\appdata\local\visualgdb\python3\lib\site-packages (from requests<3->idf-component-manager) (1.26.7)
Requirement already satisfied: idna<4,>=2.5 in c:\users\gojimmypi\appdata\local\visualgdb\python3\lib\site-packages (from requests<3->idf-component-manager) (2.7)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\gojimmypi\appdata\local\visualgdb\python3\lib\site-packages (from requests<3->idf-component-manager) (2021.10.8)
Installing collected packages: contextlib2, tqdm, schema, requests-toolbelt, click, idf-component-manager
  WARNING: The script tqdm.exe is installed in 'C:\Users\gojimmypi\AppData\Local\VisualGDB\Python3\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  Attempting uninstall: click
    Found existing installation: click 7.1.2
    Uninstalling click-7.1.2:
      Successfully uninstalled click-7.1.2
Successfully installed click-8.1.3 contextlib2-21.6.0 idf-component-manager-1.1.4 requests-toolbelt-0.9.1 schema-0.7.5 tqdm-4.64.0
WARNING: You are using pip version 21.1.1; however, version 22.1.2 is available.
You should consider upgrading via the 'C:\Users\gojimmypi\AppData\Local\VisualGDB\Python3\python.exe -m pip install --upgrade pip' command.
```

I was previously also able to get a simulated VisualGDB environment working from command-line. The sysprogs VisualGDB support recently updated [my ticket](https://sysprogs.com/w/forums/topic/esp-idf-v5/#post-32814) there (really kind, btw, considering  [... cut ...]
I'm completely confident the sysprogs folks will have a robust install _once there's an official release_. They've done a really amazing job of not only installing but also a feature: "_oh, I see there's a problem, would you like to fix it?_" that seems to work quite well.  Perhaps I just need to be patient and wait for the official release.  ;) 

But... patience is not one of my top skills and I'd really like to try out the new ESP-IDF in part to address the wolfSSL install noted in https://github.com/espressif/esp-idf/pull/9288

thanks again for taking a look, cheers :)

gojimmypi, 2022-07-28:
I've fetched the latest V5 code: `idf.py --version` reports  `ESP-IDF v5.0-dev-3954-g271f4e4c40`

I've re-[reset the python environment](https://visualgdb.com/documentation/espidf/#python) for VisualGDB. 

I've confirmed the `path/to/python/used/by/visualgdb` (in particular the "_used by VisualGDB_" part by putting in a pre-build step for `pip install --upgrade idf-component-manager`:

[image]

The `No module named 'idf_component_manager'`  is still occurring.

The first time looked somewhat promising, as indeed many dependencies were updated:
```
Executing pre-build actions
Run "pip install --upgrade idf-component-manager" in directory "C:\workspace\wolfSSL_esp32_benchmark" on local computer
pip install --upgrade idf-component-manager
Collecting idf-component-manager
  Using cached https://files.pythonhosted.org/packages/75/06/01ea2e0266111254dd4aa924c27d17b219fdb2fcae84ff7c2133f100467e/idf_component_manager-1.1.4-py3-none-any.whl
Collecting requests<3 (from idf-component-manager)
  Using cached https://files.pythonhosted.org/packages/ca/91/6d9b8ccacd0412c08820f72cebaa4f0c0441b5cda699c90f618b6f8a1b42/requests-2.28.1-py3-none-any.whl
Collecting tqdm<5 (from idf-component-manager)
  Using cached https://files.pythonhosted.org/packages/8a/c4/d15f1e627fff25443ded77ea70a7b5532d6371498f9285d44d62587e209c/tqdm-4.64.0-py2.py3-none-any.whl
Requirement already satisfied, skipping upgrade: six in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from idf-component-manager) (1.16.0)
Collecting click>=8.0.0; python_version >= "3.6" (from idf-component-manager)
  Using cached https://files.pythonhosted.org/packages/c2/f1/df59e28c642d583f7dacffb1e0965d0e00b218e0186d7858ac5233dce840/click-8.1.3-py3-none-any.whl
Requirement already satisfied, skipping upgrade: pyyaml>5.2; python_version >= "3.5" in c:\users\gojimmypi\appdata\local\programs\python\python38-32\lib\site-packages (from idf-component-manager) (6.0)
Collecting requests-toolbelt (from idf-component-manager)
  Using cached https://files.pythonhosted.org/packages/60/ef/7681134338fc097acef8d9b2f8abe0458e4d87559c689a8c306d0957ece5/requests_toolbelt-0.9.1-py2.py3-none-any.whl
Collecting schema (from idf-component-manager)
  Using cached https://files.pythonhosted.org/packages/0d/93/ca8aa5a772efd69043d0a745172d92bee027caa7565c7f774a2f44b91207/schema-0.7.5-py2.py3-none-any.whl
Collecting contextlib2>0.6.0; python_version >= "3.6" (from idf-component-manager)
  Using cached https://files.pythonhosted.org/packages/76/56/6d6872f79d14c0cb02f1646cbb4592eef935857c0951a105874b7b62a0c3/contextlib2-21.6.0-py2.py3-none-any.whl
Collecting future (from idf-component-manager)
  Using cached https://files.pythonhosted.org/packages/45/0b/38b06fd9b92dc2b68d58b75f900e97884c45bedd2ff83203d933cf5851c9/future-0.18.2.tar.gz
Collecting colorama; python_version >= "3.5" (from idf-component-manager)
  Using cached https://files.pythonhosted.org/packages/77/8b/7550e87b2d308a1b711725dfaddc19c695f8c5fa413c640b2be01662f4e6/colorama-0.4.5-py2.py3-none-any.whl
Collecting idna<4,>=2.5 (from requests<3->idf-component-manager)
  Using cached https://files.pythonhosted.org/packages/04/a2/d918dcd22354d8958fe113e1a3630137e0fc8b44859ade3063982eacd2a4/idna-3.3-py3-none-any.whl
Collecting urllib3<1.27,>=1.21.1 (from requests<3->idf-component-manager)
  Downloading https://files.pythonhosted.org/packages/d1/cb/4783c8f1a90f89e260dbf72ebbcf25931f3a28f8f80e2e90f8a589941b19/urllib3-1.26.11-py2.py3-none-any.whl (139kB)
Collecting charset-normalizer<3,>=2 (from requests<3->idf-component-manager)
  Using cached https://files.pythonhosted.org/packages/94/69/64b11e8c2fb21f08634468caef885112e682b0ebe2908e74d3616eb1c113/charset_normalizer-2.1.0-py3-none-any.whl
Collecting certifi>=2017.4.17 (from requests<3->idf-component-manager)
  Using cached https://files.pythonhosted.org/packages/e9/06/d3d367b7af6305b16f0d28ae2aaeb86154fa91f144f036c2d5002a5a202b/certifi-2022.6.15-py3-none-any.whl
Installing collected packages: idna, urllib3, charset-normalizer, certifi, requests, colorama, tqdm, click, requests-toolbelt, contextlib2, schema, future, idf-component-manager
  Running setup.py install for future: started
    Running setup.py install for future: finished with status 'done'
Successfully installed certifi-2022.6.15 charset-normalizer-2.1.0 click-8.1.3 colorama-0.4.5 contextlib2-21.6.0 future-0.18.2 idf-component-manager-1.1.4 idna-3.3 requests-2.28.1 requests-toolbelt-0.9.1 schema-0.7.5 tqdm-4.64.0 urllib3-1.26.11
WARNING: You are using pip version 19.2.3, however version 22.2.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
Run "C:\Users\gojimmypi\AppData\Local\VisualGDB\CMake\bin\cmake.exe ../../.. -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_MAKE_PROGRAM=c:/SysGCC/esp32/tools/ninja/1.10.2/ninja.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=0" in directory "C:\workspace\wolfSSL_esp32_benchmark/build/VisualGDB/Debug" on local computer
C:\Users\gojimmypi\AppData\Local\VisualGDB\CMake\bin\cmake.exe ../../.. -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_MAKE_PROGRAM=c:/SysGCC/esp32/tools/ninja/1.10.2/ninja.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=0
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32c3
-- Checking Python dependencies...
Python requirements are satisfied.
[... cut ...]
Arguments: ../../.. -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_MAKE_PROGRAM=c:/SysGCC/esp32/tools/ninja/1.10.2/ninja.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=0
Directory: C:\workspace\wolfSSL_esp32_benchmark/build/VisualGDB/Debug
Command-line action failed

========== Project Build Summary ==========
    wolfSSL_esp32_benchmark  built in 00:10
========== Build: 0 Succeeded, 1 Failed, 0 Skipped ==========

```

also tried the `python.exe` version of `pip`, with the same results:

```
Run "python.exe -m pip install --upgrade idf-component-manager" in directory "C:\workspace\EmbeddedProject1" on local computer
```

[image]

gojimmypi, 2022-12-02:
Heads up VisualGDB will soon be supporting ESP-IDF v5! Awaiting the tools to be removed from RC status.

See https://sysprogs.com/w/forums/topic/esp-idf-v5/#post-33496

georgik, 2022-12-06:
Please use the latest idf-installer to deploy 5.0 environment and test the deployment.
Feel free to re-open the issue.

@gojimmypi We can include the deployment of VisualGDB directly by online installer. We just need a command line to deploy it and URL where to download a version of the tool.

closed 2022-12-06 by georgik