espressif/idf-installer #146: last Espressif-IDE v2.6.0 with errors

edbek, 2022-08-31:
After installing Espressif-IDE v2.6.0 it is impossible to work with the product:

1. the process of downloading to Flash is not shown
2. if the volume loaded into Flash is large, then the download process is interrupted with a timeout error (attempts to fix the timeout lead to nothing). Therefore, you have to download the program separately through the terminal using "idf.py".
3. it is not possible to directly remove the flag "Flash evry time..." from the Debug Configurations. I have to remove and set other flags in parallel to disable flashing during debugging.
4. it is impossible to activate/deactivate other esp-components normally in order to integrate them into the project.

georgik, 2022-09-01:
@edbek Thank you for providing the feedback. 
@alirana01 Please, could you check what might be the source of problems with IDE?

alirana01, 2022-09-01:
@edbek Can you please respond to these

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools

**Eclipse Error log:** 
Window > Show View > Error Log

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

georgik, 2022-12-06:
Closing, because of no reaction. Feel free to reopen with requested data.

Recommended solution: upgrade to Espressif-IDE v.2.7.0 - https://github.com/espressif/idf-installer/releases/tag/espressif-ide-2.7.0

closed 2022-12-06 by georgik