espressif/idf-installer #150: IDF Installer 2.16 oes not work at all xcopy not an internal or external command

ralphde, 2022-10-22:
download from mirror slow as well. 2h for nothing.

[Setup Log 2022-10-22 #002.txt](https://github.com/espressif/idf-installer/files/9845126/Setup.Log.2022-10-22.002.txt)

georgik, 2022-11-08:
@ralphde Thank you for reporting the issue.

Please, can you check whether there is following file: `C:\Windows\system32\xcopy.exe`

ralphde, 2022-11-08:
Yes, it does exist

georgik, 2022-11-08:
Thank you for your response @ralphde . It seems that your environment variable `Path` was modified.

Please, check the environment variables in your system. Press the Windows Start logo and type `environment` - select `Edit the system environment variables`. Click `Environment Variables...` at the bottom of the `System Properties` Window. In section `System variables` find variable `Path` and it should contain `C:\Windows\System32'. If the path is missingm add it.

The result should look like:

[image]

You can test proper configuration by starting PowerShell and typing command `xcopy /?`. It should display help message to xcopy command.

ralphde, 2022-11-08:
No Sir, is and was set globally. Must have another cause!

[image: image.png]

ralphde, 2022-11-08:
> @ralphde <https://github.com/ralphde> Thank you for checking the path.
>
> Please, check whether you can run xcopy /? from the terminal.
>
> The blocking of the command might be caused by Norton Security. Please,
> check whether antivirus is not running the installation in Sandboxed
> process.
> xcopy is standard command which should be available:
> https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/xcopy
> Often antiviruses are causing problems like this.

I can run xcopy from any terminal but indeed I have Norton Antivirus

georgik, 2022-12-06:
@ralphde Please, can you check environment variable PATHEXT for the value?
It should have value like: `.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC;.CPL`

georgik, 2022-12-06:
@ralphde Problem should be solved in https://github.com/espressif/idf-installer/releases/tag/online-2.17
- added detection of PATHEXT proble in SystemCheck page
- invoked `xcopy.exe` instead of plain `xcopy`

PR #155 (2022-12-06): add offline version 5.0
Closes #110 
Closes #150
closed 2022-12-06 by georgik