espressif/idf-installer #156: esp-ide error

jinyuanqing, 2022-12-10:
when i am download,error:
```
[1/5] cmd.exe /C "cd /D D:\Espressif\frameworks\esp-idf-v4.4.3\workspace\esptimer\build\esp-idf\esptool_py && python D:/Espressif/frameworks/esp-idf-v4.4.3/components/partition_table/check_sizes.py --offset 0x8000 partition --type app D:/Espressif/frameworks/esp-idf-v4.4.3/workspace/esptimer/build/partition_table/partition-table.bin D:/Espressif/frameworks/esp-idf-v4.4.3/workspace/esptimer/build/esptimer.bin"
esptimer.bin binary size 0x29f90 bytes. Smallest app partition is 0x100000 bytes. 0xd6070 bytes (84%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D D:\Espressif\frameworks\esp-idf-v4.4.3\workspace\esptimer\build\bootloader\esp-idf\esptool_py && python D:/Espressif/frameworks/esp-idf-v4.4.3/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 D:/Espressif/frameworks/esp-idf-v4.4.3/workspace/esptimer/build/bootloader/bootloader.bin"
Bootloader binary size 0x6330 bytes. 0xcd0 bytes (11%) free.
[2/3] cmd.exe /C "cd /D D:\Espressif\frameworks\esp-idf-v4.4.3\components\esptool_py && D:\Espressif\tools\cmake\3.23.1\bin\cmake.exe -D IDF_PATH="D:/Espressif/frameworks/esp-idf-v4.4.3" -D SERIAL_TOOL="python D:/Espressif/frameworks/esp-idf-v4.4.3/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="D:/Espressif/frameworks/esp-idf-v4.4.3/workspace/esptimer/build" -P D:/Espressif/frameworks/esp-idf-v4.4.3/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32 -p COM3 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 esptimer.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.3.2
Serial port COM3
Traceback (most recent call last):
  File "D:/Espressif/frameworks/esp-idf-v4.4.3/components/esptool_py/esptool/esptool.py", line 5482, in <module>
    _main()
  File "D:/Espressif/frameworks/esp-idf-v4.4.3/components/esptool_py/esptool/esptool.py", line 5475, in _main
    main()
  File "D:/Espressif/frameworks/esp-idf-v4.4.3/components/esptool_py/esptool/esptool.py", line 4782, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "D:/Espressif/frameworks/esp-idf-v4.4.3/components/esptool_py/esptool/esptool.py", line 115, in get_default_connected_device
    _esp = chip_class(each_port, initial_baud, trace)
  File "D:/Espressif/frameworks/esp-idf-v4.4.3/components/esptool_py/esptool/esptool.py", line 345, in __init__
    self._port = serial.serial_for_url(port)
  File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM3': PermissionError(13, '拒绝访问。', None, 5)
CMake Error at run_serial_tool.cmake:56 (message):
  python
  D:/Espressif/frameworks/esp-idf-v4.4.3/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

FAILED: CMakeFiles/flash D:/Espressif/frameworks/esp-idf-v4.4.3/workspace/esptimer/build/CMakeFiles/flash 
cmd.exe /C "cd /D D:\Espressif\frameworks\esp-idf-v4.4.3\components\esptool_py && D:\Espressif\tools\cmake\3.23.1\bin\cmake.exe -D IDF_PATH="D:/Espressif/frameworks/esp-idf-v4.4.3" -D SERIAL_TOOL="python D:/Espressif/frameworks/esp-idf-v4.4.3/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="D:/Espressif/frameworks/esp-idf-v4.4.3/workspace/esptimer/build" -P D:/Espressif/frameworks/esp-idf-v4.4.3/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
Executing action: flash
Running ninja in directory d:\espressif\frameworks\esp-idf-v4.4.3\workspace\esptimer\build
Executing "ninja flash"...

```

georgik, 2022-12-22:
Hi @jinyuanqing Thank you for the error report.

I suspect that COM3 is not the port related to the ESP device. Please open Device Manager and double check whether device is recognized correctly and it's attached to the port.

jinyuanqing, 2023-01-07:
com3 is Normal and usable."serial.serialutil.SerialException: could not open port 'COM3': PermissionError(13, '拒绝访问。', None, 5)" this is a question.

georgik, 2023-01-30:
@jinyuanqing We recommend taking the following steps:
- disconnect the board from the computer
- open the command prompt and type command `mode`
- check the list of com ports. None of these is used for ESP device
- connect the board
- type the command again: `mode`
- new COM port should appear (or two)
- flash the project to the COM port

If failure still happens, please open Device Manager and check for any signs of a misconfigured device. If you discover one, remove it from Device Manager, including drivers. Run idf-installer to reinstall drivers https://github.com/espressif/idf-installer#manual-installation-of-drivers. Plug in the device.

If the problem persists, use [UsbTool](https://visualgdb.com/UsbDriverTool/) to switch driver.

If the problem persists, use a different USB cable and a different USB port.

If the problem persists, use a different ESP32 board, ideally from a different batch.

If the problem persists, use a different computer.

Feel free to ask further questions related to flashing at https://esp32.com

closed 2023-01-30 by georgik