espressif/idf-installer #159: When I use offline to install IDFv5, It need download with github

WangShuoran, 2022-12-24:
I am forget screenshot, but when I install to something seem with riscv, it will need download with github.

igrr, 2022-12-24:
Please attach the installer log file, it can be found in your profile's temporary directory.

WangShuoran, 2022-12-24:
I am use powershell `certutil.exe -hashfile .\esp-idf-tools-setup-offline-5.0.exe SHA1` to output hash is "6162696a9803e18b67ca1bea3040560f4ffb6d1f" it is ok. but I have another version with SHA1 is “cb9342ea8ee52edd296d10ec0c72269f2c8c47a9” which will download:
[output.txt](https://github.com/espressif/idf-installer/files/10298259/output.txt)

WangShuoran, 2022-12-24:
I confirm I download those at your official website, I don't know why I have last two version:
[image]

I suggest give exe installer an SHA1 let us check, Thank you.

igrr, 2022-12-24:
The newer one you have probably includes the fix for the missing tools issue, which was merged yesterday: https://github.com/espressif/idf-installer/pull/158

WangShuoran, 2022-12-24:
Thank you. But I still suggest give exe installer an SHA1 let us check.

closed 2022-12-24 by WangShuoran