espressif/idf-installer #160: Espressif-IDE 2.7.0 always download crosstool-NG release file from github (IEP-844)

mzhboy, 2023-01-02:


igrr, 2023-01-02:
This is probably because we also [require](https://github.com/espressif/esp-idf/blob/0025915dc489a9d45f99aed74920346f8ac4ec09/tools/tools.json#L453-L454) riscv32-esp-elf-gcc for ESP32-S2 and ESP32-S3 (due to their RISC-V ULP coprocessors), but the installer currently [doesn't install](https://github.com/espressif/idf-installer/blob/8c88b18e528ec3b4d2898cdaecad918aaa4bf5b6/src/InnoSetup/IdfToolsSetup.iss#L241-L254) `riscv32-esp-elf` tool for ESP32-S2 and ESP32-S3.

(This was also mentioned in https://github.com/espressif/idf-installer/pull/158#discussion_r1055237061.)

mzhboy, 2023-01-02:
anyway, I manage finish install ESP-IDE by 
 1. download `xpack-riscv-none-elf-gcc-12.2.0-1-win32-x64.zip` and put it into `Espressif\dist\`
 2. select [esp32-c3] (basically leave everything default)

I don't konw which option works, but it worked.

mzhboy, 2023-01-02:
I  highly suggest download extra package from espressif.com or make `offline installer` truly `offline`, we don't really mind a bigger installer package.

georgik, 2023-01-11:
@mzhboy There are updated installers available:
- https://github.com/espressif/idf-installer/releases/tag/espressif-ide-2.7.0-esp-idf-4.4.3
- https://github.com/espressif/idf-installer/releases/tag/espressif-ide-2.7.0-esp-idf-5.0

Please test the installation.

PR #161 (2023-01-13): Bugfix/missing risc in s2 s3
Closes: #160
closed 2023-01-13 by georgik