espressif/idf-installer #172: Installs old openocd-esp32 v0.11.0-esp32-20221026 that doesn't support esp32c6

RJSDevel, 2023-03-01:
Installer 2.20 installs old openocd-esp32 v0.11.0-esp32-20221026 that doesn't support esp32c6

igrr, 2023-03-01:
Hi @RJSDevel,

ESP32-C6 support in IDF is still in progress. As soon as OpenOCD version is updated [here](https://github.com/espressif/esp-idf/blob/94c87a91060ff6e5962ec19e7152fc402e6c704c/tools/tools.json#L771), the installer will start downloading the latest release.

The reason IDF hasn't been updated to depend on the latest OpenOCD release is that we found an issue in that release and will prepare a new one in the next few days. Afterwards we will update the dependency.

RJSDevel, 2023-03-01:
> Hi @RJSDevel,
>
> ESP32-C6 support in IDF is still in progress. As soon as OpenOCD version is updated [...]

Could you say what the issues? Because I've just taken the last release and put it into v0.11.0-esp32-20221026 folder and openOCD recognize the esp32c6 but debbuging isn't working via builtin USB-JTAG 

Log 

Open On-Chip Debugger v0.12.0-esp32-20230221 (2023-02-21-11:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Flashing C:/Users/Developer/workspace/HA_on_off_light/build/bootloader/bootloader.bin at 0x0
** program_esp input args <0x0 verify> **
Info : esp_usb_jtag: serial (60:55:F9:F6:F8:34)
Info : esp_usb_jtag: Device found. Base speed 24000KHz, div range 1 to 255
Info : clock speed 24000 kHz
Info : JTAG tap: esp32c6.cpu tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 2 harts
Info :  hart 0: XLEN=32, misa=0x40903105
Info : starting gdb server for esp32c6 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c6.cpu tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : Reset cause (24) - (JTAG CPU reset)
Info : [esp32c6] Found 4 triggers
Info : Flash mapping 0: 0x10020 -> 0x42060020, 50 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 347 KB
Info : Auto-detected flash bank 'esp32c6.flash' size 8192 KB
Info : Using flash bank 'esp32c6.flash' size 8192 KB
** Programming Started **
Info : PROF: Erased 24576 bytes in 399.005 ms
Info : PROF: Data transferred in 287.001 ms @ 83.6234 KB/s
Info : PROF: Wrote 24576 bytes in 706.994 ms (data transfer time included)
** Programming Finished in 1996 ms **
** Verify Started **
Info : PROF: Flash verified in 220.002 ms 
** Verify OK **
** Flashing done for bootloader/bootloader.bin in 2216 ms **
Flashing C:/Users/Developer/workspace/HA_on_off_light/build/light_bulb.bin at 0x10000
** program_esp input args <0x10000 verify> **
Info : JTAG tap: esp32c6.cpu tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : Reset cause (24) - (JTAG CPU reset)
** Programming Started **
Info : PROF: Erased 466944 bytes in 1491.99 ms
Info : PROF: Data transferred in 5342 ms @ 85.3613 KB/s
Info : PROF: Wrote 466944 bytes in 5808 ms (data transfer time included)
** Programming Finished in 7610 ms **
** Verify Started **
Info : PROF: Flash verified in 455.994 ms 
** Verify OK **
** Flashing done for light_bulb.bin in 8072 ms **
Flashing C:/Users/Developer/workspace/HA_on_off_light/build/partition_table/partition-table.bin at 0x8000
** program_esp input args <0x8000 verify> **
Info : JTAG tap: esp32c6.cpu tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : Reset cause (24) - (JTAG CPU reset)
** Programming Started **
Info : PROF: Erased 4096 bytes in 242.999 ms
Info : PROF: Data transferred in 64.001 ms @ 62.499 KB/s
Info : PROF: Wrote 4096 bytes in 463 ms (data transfer time included)
** Programming Finished in 1016 ms **
** Verify Started **
Info : PROF: Flash verified in 215 ms 
** Verify OK **
** Flashing done for partition_table/partition-table.bin in 1231 ms **
** Total programming time 11520 ms **
** Resetting Target **
Info : JTAG tap: esp32c6.cpu tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : Reset cause (24) - (JTAG CPU reset)
shutdown command invoked

igrr, 2023-03-01:
It's a bit hard to say, this log doesn't show any errors, just a successful sequence of flashing the device.

If you have a problem with debugging, please open an issue at https://github.com/espressif/openocd-esp32/issues/ and describe in detail how you are launching OpenOCD and the issue you are seeing. You could also try to launch the tool manually from the command line to figure out if it's an openocd issue or an Eclipse IDE issue. (Both are possible at this point, since C6 support is pretty new and not all the tools are up to date yet.)

georgik, 2023-03-31:
Installer 2.21 with `master` branch which has preliminary support of C6 now deploys `v0.12.0-esp32-20230313`.
Thank you for reporting the issue.
Feel free to reopen in case of problems.

closed 2023-03-31 by georgik