espressif/idf-installer #182: Internet for offline installer

Yurbaan, 2023-04-12:
Hello everyone!
I encountered such an error when using esp-idf 5.0.1 offline installer (for Windows). Do I understand correctly that an internet connection is needed for offline installation?:)
```
                          Downloading https://github.com/espressif/esp-rom-elfs/releases/download/20220823/esp-rom-elfs-20220823.tar.gz
                          Destination: C:\Espressif\dist\esp-rom-elfs-20220823.tar.gz.tmp
                          WARNING: Download failure: <urlopen error [Errno 11001] getaddrinfo failed>
                          WARNING: Failed to download https://github.com/espressif/esp-rom-elfs/releases/download/20220823/esp-rom-elfs-20220823.tar.gz to C:\Espressif\dist\esp-rom-elfs-20220823.tar.gz.tmp
                          Downloading https://github.com/espressif/esp-rom-elfs/releases/download/20220823/esp-rom-elfs-20220823.tar.gz
                          Destination: C:\Espressif\dist\esp-rom-elfs-20220823.tar.gz.tmp
                          WARNING: Download failure: <urlopen error [Errno 11001] getaddrinfo failed>
                          WARNING: Failed to download https://github.com/espressif/esp-rom-elfs/releases/download/20220823/esp-rom-elfs-20220823.tar.gz to C:\Espressif\dist\esp-rom-elfs-20220823.tar.gz.tmp
                          Downloading https://github.com/espressif/esp-rom-elfs/releases/download/20220823/esp-rom-elfs-20220823.tar.gz
                          Destination: C:\Espressif\dist\esp-rom-elfs-20220823.tar.gz.tmp
                          WARNING: Download failure: <urlopen error [Errno 11001] getaddrinfo failed>
                          WARNING: Failed to download https://github.com/espressif/esp-rom-elfs/releases/download/20220823/esp-rom-elfs-20220823.tar.gz to C:\Espressif\dist\esp-rom-elfs-20220823.tar.gz.tmp
                          ERROR: Failed to download, and retry count has expired
                          Please make sure you have a working Internet connection.
                          By downloading and using the offline installer from https://dl.espressif.com/dl/esp-idf you might be able to work around this issue.
```
Log after error

georgik, 2023-04-28:
Thank you for reporting the problem. 

We're investigating why those tools were not downloaded during build process.

Also CI scripts for Offline testing in Hyper-V and Azure will be added.

georgik, 2023-04-28:
Note from research: It's possible to turn off the network even in Windows Sandbox

georgik, 2023-04-28:
Thank you @Yurbaan for pointing at the problem. We found that also some other files from `tools` directory were not included.
We yanked the 5.0.1 release and re-run the build with proper configuration to include also missing files.
The new rebuild of installer is available at https://github.com/espressif/idf-installer/releases/tag/offline-5.0.1
Please, test the updated version.

kumekay, 2023-04-29:
@georgik

The component manager supports two essential features required for the desired behavior: it can utilize a registry mirror, and the registry mirror can be a local directory. Hypothetically, we could include a partial registry mirror in the offline installer.

However, some limitations need to be addressed first:
- There is no straightforward method for creating a partial mirror of the registry, containing only components used in examples.
- The component manager expects a full mirror and will attempt to fetch all required components for a project from it. Consequently, if you add components to your project that are not available in the partial mirror, you must switch to a full mirror.
- Components used in examples have independent development cycles and are not strictly tied to specific ESP-IDF versions. Relying on an offline mirror could lead to the usage of dependencies with bugs and security vulnerabilities.

We must first determine how to mitigate these issues and ensure that the behavior in offline mode is not confusing.

P.S.: I have fixed the mixed slashes issue, and it will be included in the next release of the component manager

georgik, 2023-05-03:
Thank you @kumekay for the explanation.

Basically, examples which are using components introduced in 5.x requires connection to the internet in build time.

The build with components could work offline to some extend. For this release, we declare it as a limitation. The limitation might be removed in future releases after updating the implementation of Component Manager.

The installation works completely offline and installer is not failing.

PR #184 (2023-07-20): Test/release 2.22
Closes #182 
Closes #188
closed 2023-07-20 by georgik