espressif/idf-installer #186: The ESP-IDF path is not installed in the directory which I selected

Rainbow-Cai, 2023-05-22:
I used the online installer package to install the Master version of the esp-idf SDK

When I select to install the Master version of esp-idf SDK in my "F:\esp" directory, it will automatically install in my "E:\esp" directory during the installation process. It will installing the "Espressif" files in both my "E:\esp" and "F:\esp" directories at the same time.

As a result, when I install the esp-idf SDK of v5.0 version in the "E:\esp" directory , I cannot use the esp-idf SDK of Master version in   "F:\esp" directory at the same time. 

My directory structure is as follows:

E:\esp
  - esp-idf-v5.0
  - Espressif

  
F:\esp
  - esp-idf-master
  - Espressif
[image]

The error log is in the  attachment.
[Setup Log 2023-05-22 #009.txt](https://github.com/espressif/idf-installer/files/11528803/Setup.Log.2023-05-22.009.txt)

georgik, 2023-05-23:
Thank you @Rainbow-Cai  for reporting the issue. The installation consists of two parts:
- deployment of tools - default location `C:\Espressif\` + tools
- deployment of a framework - default location `C:\Espressif\framework\esp-idf-...`

Note: It's possible to deploy multiple instances of ESP-IDF. It's not necessary to deploy tooling twice. You can pick the directory for tools from E and then deploy new instance of esp-idf to F.

Just to be sure that I understand the problem well:
- you have following drives on your computer: C, E, F
- you have no installation of ESP-IDF before (no environment variable like IDF_TOOLS_PATH set) 
- you select installation on F
- the deployment deploys both to E and F

Rainbow-Cai, 2023-05-24:
Dear @georgik  So, can't I deploy the  different  version SDK tool  in the different folder? 
I hope different SDK versions have separate its toolchain.

georgik, 2023-05-24:
@Rainbow-Cai There is no need to deploy different tooling to different directories.
Initialization scripts of each ESP-IDF are adding correct toolchains for each version of ESP-IDF. So the tooling can be deployed to the same container directory.
ESP-IDF is reading tools.json during initialization to determine the proper version: https://github.com/espressif/esp-idf/blob/master/tools/tools.json
If the toolchain contains the same tools for different versions of ESP-IDF, it will be automatically reused.

leeebo, 2023-05-24:
@georgik Thanks for your detailed explanation, but to avoid user confusion, it's better to be consistent with user expectations. 

If users chose another tools path when they install the second ESP-IDF instance, It is better to install it where the user wants, or guide the customer to an existing tools path and tell the reason (eg. Found an existing tool path. Strongly recommended...)