espressif/idf-installer #187: Trying to install on a Flash drive : Installation has failed with exit code 128

georgewicks, 2023-05-29:
Used latest online installer

Slightly different error from the install on my setup:
[Setup Log 2023-05-29 #001.txt](https://github.com/espressif/idf-installer/files/11594327/Setup.Log.2023-05-29.001.txt)

georgik, 2023-05-30:
@georgewicks Thank you for reporting the error.

Installing on flash drive is interesting option. Yet from performance point of view we would recommend to deploy the installation on SSD drive.

The problem might be connected with FAT filesystem and git meta atributes. Which filesystem do you have on flash drive? NTFS or FAT?

As a workaround you can try Offline installer, which does not perform the problematic git operation.
https://github.com/espressif/idf-installer/releases

georgik, 2023-05-30:
The problem seems to be caused by "xcopy" which is not transferring to FAT-fs in the case of Online installer when the repo is downloaded as ZIP instead of cloning, which is the situation for the release version.

In the case of Offline installer, we're changing attributes of some directories, so the Windows does not have a problem with it using `attrib` command. This approach works well: https://github.com/espressif/idf-installer/blob/main/Build-Installer.ps1#L254

One possible correction would be to include this change also after extracting ZIP file.

We can also confirm that using Offline installer is a working solution even for FAT filesystem.

georgik, 2023-07-20:
The attempt to remove xcopy from the installer failed. The main reason is that 7zip is not able to skip top level directory when extracting archive. It will require different extraction mechanism, so that ESP-IDF is stored in correct location.
Moving to 2.23