espressif/idf-installer #204: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND

dobairoland, 2023-10-25:
I've just used the 2.23 version of the online installer, picked the "existing ESP-IDF" and Eclipse IDE options.

I'm experiencing the following issue both in command line and Eclipse:

```
C:\Users\Roland\esp\esp-idf\examples\get-started\blink>idf.py openocd
C:\Users\Roland\esp\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: openocd
Note: OpenOCD cfg not found (via env variable OPENOCD_COMMANDS nor as a --openocd-commands argument)
OpenOCD arguments default to: "-f board/esp32-wrover-kit-3.3v.cfg"
OpenOCD started as a background task 8844
Executing action: post_debug
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND
Error: unable to open ftdi device with description '*', serial '*' at bus location '*'
C:\Users\Roland\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:4: Error:
at file "C:\Users\Roland\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 4
"openocd" exited with 1
```

dobairoland, 2023-10-25:
@jakub-kocka PTAL

jakub-kocka, 2023-10-26:
@dobairoland, it is Windows specific driver issue. There are installed most common drivers for platforms and Windows is using for some historic reasons theirs approach.

I was able to replicate and solved it thanks to Tomas R.
This should help: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html?highlight=zadig#windows

IIUC this has to be handled manually because driver change can mess more things than improve.

But as we agreed with Tomas, we can definitely try to catch this Error and add a hint to point the user for solution.

dobairoland, 2023-10-27:
@jakub-kocka Yes, thanks. I'm aware of the manual way using Zadig. It was the only way for a long time. 

However, now https://github.com/espressif/idf-env should have been used with the installer. Are you able to reproduce and fix with that one as well?

I have kept my computer broken and didn't try Zadig so we can get out of it some information regarding the failure (if needed).

dobairoland, 2023-10-30:
@alirana01 JFYI, this is the issue from the Espressif-IDE Workshop.

jakub-kocka, 2024-07-30:
This has been solved by the [WROVER KIT driver](https://github.com/espressif/esp-win-usb-drivers/releases/tag/ESP-WROVER-KIT_v1.0.0.0) in the [esp-win-usb-drivers repository](https://github.com/espressif/esp-win-usb-drivers).

closed 2024-07-30 by jakub-kocka