espressif/idf-installer #206: Add websocket python package in the windows offline installer

kolipakakondal, 2023-10-30:
Please include the WebSocket Python package in the Windows offline installer, as it will be required by the serial monitor; otherwise, it will fail if someone working without internet connection.

This was handled in the IDE during the tools installation process, however but that requires internet again!

dobairoland, 2023-11-03:
One way to address this is to add a new IDE "feature" to ESP-IDF. It would be a new file in https://github.com/espressif/esp-idf/tree/master/tools/requirements where you could place all additional IDE requirements and eliminate to handle this at the IDE plugin side. Then the installer could just run the following (if IDE support was selected):

```
idf_tools.py install --features=core,ide
```

For CLI-only environments these packages won't be installed.

This would also ensure that the packages are tested to be compatible with other ESP-IDF requirements and issues like https://github.com/espressif/vscode-esp-idf-extension/issues/1054 would not happen.

kolipakakondal, 2024-05-15:
> One way to address this is to add a new IDE "feature" to ESP-IDF. It would be a new file in https://github.com/espressif/esp-idf/tree/master/tools/requirements where you could place all additional IDE requirements and eliminate to handle this at the IDE plugin side. Then the installer could just run the following (if IDE support was selected):
> 
> ```
> idf_tools.py install --features=core,ide
> ```
> 
> For CLI-only environments these packages won't be installed.
> 
> This would also ensure that the packages are tested to be compatible with other ESP-IDF requirements and issues like [espressif/vscode-esp-idf-extension#1054](https://github.com/espressif/vscode-esp-idf-extension/issues/1054) would not happen.

That's good advice for managing the extra dependencies required by IDEs. I somehow lost this thread. We can create a requirements.ide.txt file and include [websocket-client](https://pypi.org/project/websocket-client/) as a dependency.