espressif/idf-installer #231: Targets like esp32s3 do not appear in Espressif IDE after using Online Installer 2.24

crispmfc, 2024-03-21:
I have reinstalled 5 times activating only ESP32-S3, ESP32-S2 and ESP32-S3, ESP32 and ESP32-S3, all ESP32 series and whatever the case, it is always the same result: in IDF target menu of Espressif IDE they are only available: esp32 esp32c2 esp32c3 esp32h2. See screenshot 3.

Universal Online Installer 2.24
ESP-IDF 5.2.1
Windows Defender. I don't think it's an antivirus problem. I tried the installation with windows defender enabled, disabled, I also enabled the option to add IDF to the exclusion list.

Edition	Windows 11 Pro
Version	23H2
Installed on	‎3/‎10/‎2024
OS build	22631.3296
Experience	Windows Feature Experience Pack 1000.22687.1000.0

[image]
[image]
[image]

crispmfc, 2024-03-21:
Fortunately the offline installer "Espressif-IDE 2.12.0 with ESP-IDF v5.1.2 - Offline Installer" does not have this problem.
[image]

crispmfc, 2024-03-21:
The best workaround I found was to install ESP-IDE + IDF 5.2 offline "beta" from here:
"[Release of Espressif IDE 2.12.1 with ESP-IDF 5.2](https://github.com/espressif/idf-installer/releases/tag/espressif-ide-2.12.1-esp-idf-5.2)"
[image]

copych, 2024-06-19:
Windows 10. Clean installed IDE, using online installer, chosen master branch of ESP-IDF, cause I wanted latest changes in peripherals. Now I have the same issue with absent targets. Prior to that I have used 4.4.2 if I remember correctly, and there were targets, and I was able to compile and run projects.