espressif/idf-installer #269: Installation has failed with exit code 101 - missing ftdi

kzajac83, 2024-08-29:
Hello

The same problem as #100 

[image]

Chossing all option like zip archives or githubs repositoers nothing change.

[image]

Genere missing files mentioned in error log but I try  install as admin but not help too. 

Help only  exclude ftdi drivers from installer. 

My OS in Windows 10 Home

PS. On all Windows channel button now work after appear error, eg clicking cancel here nothing happened:

[image]

kzajac83, 2024-08-29:
@kehkou Why need download  something from mediafire.com, some mailware probabky this is

kzajac83, 2024-08-29:
PS. On installing occure error on end, not sure finally installed or not?? 

[image]

[Setup Log 2024-08-29 #004.txt](https://github.com/user-attachments/files/16795656/Setup.Log.2024-08-29.004.txt)

jakub-kocka, 2024-08-30:
Hello @kzajac83, we are sorry for the inconvenience there has been some issue with the builds of the Installer because of the FTDI driver.

Right now all the builds should be fixed again and available on the Installer's page https://dl.espressif.com/dl/esp-idf/

Thank you for your understanding.

kzajac83, 2024-08-30:
Thank you @jakub-kocka for driver issue fix however 2nd issue still exist:

[image]

[Setup Log 2024-08-30 #002.txt](https://github.com/user-attachments/files/16812917/Setup.Log.2024-08-30.002.txt)

jakub-kocka, 2024-08-30:
@kzajac83 Can you try to install the IDF into the `C` drive?
This error looks a bit strange and this might help to move.

kzajac83, 2024-08-30:
Installing 1st time [esp-idf-tools-setup-online-2.28.exe](https://dl.espressif.com/dl/idf-installer/esp-idf-tools-setup-online-2.28.exe?) (Universal Installer) appear request to add something to MS registry and I do that, next appear issue with drivers plus 2nd mentioned issue.  Next time using Universal Installer no request for add key to the registry and after add missing drivers stay only 2nd issue. Before every next attempt installation deleting D:\Espressif directory.

Next I decide to download ~2GB offline installer [esp-idf-tools-setup-espressif-ide-3.0.0-with-esp-idf-5.2.2.exe](https://dl.espressif.com/dl/idf-installer/esp-idf-tools-setup-espressif-ide-3.0.0-with-esp-idf-5.2.2.exe?) and on begging of installation appear request to add below to registry:

`powershell -Command "&{ Start-Process -FilePath reg 'ADD HKLM\SYSTEM\CurrentControlSet\Control\FileSystem /v LongPathsEnabled /t REG_DWORD /d 1 /f' -Verb runAs}"
`

I don't know this is the same as Universal Installer request one time but, if yest then I add it already and Universal Installer not requested any more for that.  But Offline Installer request the same (or this is other registry requesty add ??).

Anyway after use Offline Installer and use drive D:\ all pass fine. 

Something wrong must be in Universal Installer, maybe something is not downloading? 

Anyway probably I will not more need installing hence cannot help more but if something need please ask. 

BR, Krzysiek

jakub-kocka, 2024-08-30:
Thank you for the information, we will take a look and check if everything is correct.

I am glad that it works for you.

jappies, 2024-12-16:
Seems that the FTDI driver download is not working

[image]

georgik, 2024-12-16:
@hahihula Would it be possible to use newly developed Insaller to resolve the problem?