espressif/idf-installer #278: Can't install on Windows 11 ARM64

TMumbai, 2024-09-29:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Please support Windows 11 ARM64. When I try to install, I get the error message this windows version is not supported.

### Use cases

I believe all windows 11 machine moving forward will be ARM64 so appreciated this feature

### Alternatives

Using WSL but dont want to use it if possible and use Windows 11 natively

dobairoland, 2024-09-30:
cc @Hahihula for the feature request for idf-im. Is this already supported or should I create a task there?

jakub-kocka, 2024-10-07:
We are closing this since it is not planned for this version of Installer. We will consider this in the new one as mentioned above.

closed 2024-10-07 by jakub-kocka