espressif/idf-installer #305: Unable to install 'esp-idf-tools-setup-online-2.3.4' and others (IDFGH-14606)

carlessole, 2025-02-09:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The installation should succeed.

### What is the actual behavior?

Installation fails with the following error:

> ERROR: Failed to check the tool while installed. Removing directory C:\Espressif\.espressif_v5.4\tools\ccache\4.10.2

Sometimes it fails at another step. But is always failing

### Steps to reproduce.

Try to install ESP-IDF with the latest installer on Windows 11

### Build or installation Logs.

```plain
See full log on attached file
```

### More Information.

I have also try it disabling completely the McAfee antivirus and Windows Defender. Error is still there

[Setup Log 2025-02-09 espressif/esp-idf#1.txt](https://github.com/user-attachments/files/18722531/Setup.Log.2025-02-09.001.txt)

carlessole, 2025-02-10:
If I try to install it from the Espressif-IDE, I got the error at the same point, but it gives me more info:
```
Extracting C:\Espressif\.espressif_v5.4\dist\ccache-4.10.2-windows-x86_64.zip to C:\Espressif\.espressif_v5.4\tools\ccache\4.10.2
ERROR: tool ccache version 4.10.2 is installed, but getting error: non-zero exit code (3) with message: terminate called after throwing an instance of 'std::filesystem::__cxx11::filesystem_error'

  what():  filesystem error: Cannot convert character sequence: Illegal byte sequence

ERROR: Failed to check the tool while installed. Removing directory C:\Espressif\.espressif_v5.4\tools\ccache\4.10.2
```

```
ERROR: tool ccache has no installed versions. Please run 'C:\Program Files\Python313\python.exe C:\Espressif\frameworks\esp-idf_v5.4\tools\idf_tools.py install' to install it.
ERROR: tool dfu-util has no installed versions. Please run 'C:\Program Files\Python313\python.exe C:\Espressif\frameworks\esp-idf_v5.4\tools\idf_tools.py install' to install it.
ERROR: tool esp-rom-elfs has no installed versions. Please run 'C:\Program Files\Python313\python.exe C:\Espressif\frameworks\esp-idf_v5.4\tools\idf_tools.py install' to install it.
```

mfialaf, 2025-02-10:
Hi @carlessole, I've installed esp-idf using the online installer without any issues.

The issue `Illegal byte sequence`, seems to be some kind of encoding style error. I've checked the ccache github repository and found this issue: https://github.com/ccache/ccache/issues/1409.

To isolate the issues, can you try to download the release (https://github.com/ccache/ccache/releases/tag/v4.10.2) and run the commands from their issue report?
The same release is downloaded by the esp-idf-online-installer (as can be seen in yours attached log) in `C:\Espressif\.espressif_v5.4\dist\ccache-4.10.2-windows-x86_64.zip` and extracted to `C:\Espressif\.espressif_v5.4\tools\ccache\4.10.2`

We are using their latest release `4.10.2`.

I can see that the fix commit (from the issue) is planned for version `4.11`. But as the 4.11 is not released yet, there is nothing we can do about it right now.

Please let me know if you made any porgress or tested the ccache without esp-idf.