espressif/idf-installer #312: Installation failed WIN11 ARM64 ESP-IDF 5.4 (IDFGH-14764)

RenaudC82, 2025-03-03:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF V5.4 offline installer

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I'm expect to install  esp-idf-tools-setup-offline-5.4 on windows 11 24h2 with an ARM64 processor

### What is the actual behavior?

After running the exe : esp-idf-tools-setup-offline-5.4 ans selecting the languuage i a have the following eroor message : "This program does not support the version of Windows your computer is running"

[image]

### Steps to reproduce.

1. Use a laptop win WIN 11 24H2 and a Snapdragon X elite processor 
2. double click on the executable file
3. select the language
4. error message appear

### Build or installation Logs.

```plain

```

dobairoland, 2025-03-03:
Hi @RenaudC82. ARM64 on Windows is not supported. We are tracking this request but it won't happen in this installer. It is deprecated and soon will be replaced with a new installer.

dobairoland, 2025-03-03:
cc @Hahihula

RenaudC82, 2025-03-05:
Thanks for the info.

spirilis, 2026-01-07:
The eim-CLI for Windows x64 seems to run via PowerShell for me (Windows 11, Surface Pro 11 ARM64).

Releases: https://github.com/espressif/idf-im-ui/releases

However, the install did not complete (some kind of conflicting dependency error).  At least this isn't an architecture-specific issue.

jakub-kocka, 2026-01-07:
I can see that you have opened a ticket in the EIM repository @spirilis. https://github.com/espressif/idf-im-ui/issues/328
EIM should be the target installer for the Windows ARM.
I am closing this tracker as "not planned" because, as it has been said, this installer is not going to get Windows ARM support and will become a legacy one starting ESP-IDF v6.0.0.

basem357, 2026-02-05:
Really need to have esp idf on windows 11 arm. any news ?

Hahihula, 2026-02-09:
> > > Really need to have esp idf on Windows 11 arm. any news ?
> > 
> > 
> > Hello [@basem357](https://github.com/basem357), as mentioned above, this Installer most probably won't be getting Windows ARM support. Please check EIM - https://github.com/espressif/idf-im-ui
> > Or you can try to use [WSL](https://learn.microsoft.com/en-us/windows/wsl/install) for now
> 
> Thanks for the reply, with eim I can't figure out which installer to use, there is no win arch64. With x64 the installation gets stuck at the phyton step, it says it needs to be installed but it doesn't install it.

Hi @basem357,
currently only supported approach for arm Windows is to use eim inside the wsl. Use ubuntu and the linux Linux ARM64 flavor of EIM.