espressif/idf-installer #313: Installing ESP IDF tools fails on windows 10 PC (IDFGH-14784)

hitecSmartHome, 2025-03-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-tools-setup-offline-5.4

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I expected it to be installed

### What is the actual behavior?

It crashes and exists with code 1

### Steps to reproduce.

1. Get a windows 10 pc
2. Download `esp-idf-tools-setup-offline-5.4`
3. Try to install it
4. Crash
5. Profit?

### Build or installation Logs.

```plain
Logs are huge. I'm uploading it as a file.
```

### More Information.

I have tried to install it also on C into it's separate folder.
The thing is that I'M already using esp idf with vscode but I need a separate installation to debug coreDump file which is extracted from my esp.

[image]

dobairoland, 2025-03-06:
Hi @hitecSmartHome.

> Logs are huge. I'm uploading it as a file.

Where is that file? I'm sorry but we cannot help based on the screenshot.

> but I need a separate installation to debug coreDump file

In that scenario you don't need a completely new installation. You can manually install esp-gdb and esp-coredump tools. More information can be found at https://github.com/espressif/esp-coredump.

hitecSmartHome, 2025-03-07:
Sorry, it did not upload it for some reason

hitecSmartHome, 2025-03-11:
I have tried to install it again today and it installed. I don't know what changed. My antivirus is still there. I just redownloaded the offline installer again. Thanks for the help!

closed 2025-03-11 by hitecSmartHome