espressif/idf-installer #329: Configuration Error With ESP-IDF in VSCode Extension - No module named pip

KirinRyuuri, 2025-07-28:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

VSCode 的 ESP-IDF 插件能一路“下一步”安装完成。

### Actual behavior (suspected bug)

提示 "No module named pip"

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

1. 使用 Universal Online Installer 2.3.5 安装完成
2. 在 VSCode 中安装 ESP-IDF 插件
3. 在配置 ESP-IDF 插件的时候，提示 `C:\Espressif\tools\idf-python\3.11.2\python.exe: No module named pip`。

### Project release version

Universal Online Installer 2.3.5

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

other (details in Additional context)

### Additional context

4. 手动执行 `C:\Espressif\tools\idf-python\3.11.2\python.exe -m ensurepip` 安装pip
5. 重新配置 ESP-IDF 插件，可以配置完成。

注：如果执行 `C:\Espressif\tools\idf-python\3.11.2\python.exe -m pip` 会提示 `No module named pip`。

jakub-kocka, 2025-07-28:
Hi, @KirinRyuuri,

We are sorry for the inconvenience.
First of all, using the ESP-IDF installer and the VS Code Extension are two different things. With the Installer, you can start using ESP-IDF via CLI after the installation.
I've tested your issue with the plain Windows installation, and I can reproduce it.

If you have used Universal Online Installer 2.3.5 to install ESP-IDF, then in VS Code's ESP-IDF Extension, you can choose the "USE EXISTING SETUP" option during configuration, which should work just fine for now.

Adding ESP-IDF Extension documentation: https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/

jakub-kocka, 2025-08-04:
Hi, @KirinRyuuri,
Yes, it should be in another repository.

However, thanks for pointing this out. There has already been an open PR, which should solve the issue.

https://github.com/espressif/vscode-esp-idf-extension/pull/1625

I am closing this issue, since it is not related to this repository.